markniu/Bed_Distance_sensor #165: Trouble adding to ramps1.4 board

peterstevey, 2024-07-16:
I've added the bd sensor to my project build but having issues getting the sensor to work correctly. 

M102 S-1 doesn't return as expected (no mention of V1.0 pandapi3d.com):

>>> M102 S-1
SENDING:M102 S-1
ok

Calibration works fine, and the mesh is saved without problem. 

Endpoint is not triggered, and distance is not displayed on LCD. 

Software is configured exactly as outlined in Chris's Basement tutorial and on the github documentation.

Changing pins doesnt seem to help. Would updating the bdsensor firmware affect this? 

Calibration data seems good, although have little benchmark to compare to. 
SENDING:M102 S-5
Calibrate data:0,202, check:1
Calibrate data:1,229, check:1
Calibrate data:2,258, check:1
Calibrate data:3,287, check:1
Calibrate data:4,318, check:1
Calibrate data:5,347, check:1
Calibrate data:6,377, check:1
Calibrate data:7,407, check:1
Calibrate data:8,438, check:1
Calibrate data:9,468, check:1
Calibrate data:10,497, check:1
Calibrate data:11,524, check:1
Calibrate data:12,553, check:1
Calibrate data:13,579, check:1
Calibrate data:14,606, check:1
Calibrate data:15,628, check:1
Calibrate data:16,653, check:1
Calibrate data:17,676, check:1
Calibrate data:18,696, check:1
Calibrate data:19,717, check:1
Calibrate data:20,738, check:1
Calibrate data:21,756, check:1
Calibrate data:22,774, check:1
Calibrate data:23,792, check:1
Calibrate data:24,809, check:1
Calibrate data:25,824, check:1
Calibrate data:26,838, check:1
Calibrate data:27,853, check:1
Calibrate data:28,867, check:1
Calibrate data:29,879, check:1
Calibrate data:30,890, check:1
Calibrate data:31,901, check:1
Calibrate data:32,914, check:1
Calibrate data:33,924, check:1
Calibrate data:34,933, check:1
Calibrate data:35,942, check:1
Calibrate data:36,950, check:1
Calibrate data:37,1015, check:1
Calibrate data:38,1015, check:1
Calibrate data:39,1015, check:1

markniu, 2024-07-16:
how about increase the delay value `#define  I2C_BD_DELAY  20`  from 20 to 40  ?  
and please use this firmware that I pulled to official marlin recently. https://github.com/markniu/MarlinPULL/tree/bugfix-2.1.x
https://github.com/MarlinFirmware/Marlin/pull/27243

peterstevey, 2024-07-16:
I have tried increasing the delay, but doesnt change it. 

I will recompile with the bugfix code and see if that solves it. Thank you!

peterstevey, 2024-07-18:
Sorry - I have another issue now... 

Homing works well, and z height is accurate! 

When initiating G29, the axis are homed, but then marlin crashes and the printer restarts. Sometimes it beeps and just reports "error" on screen... 

ABL is activated, probing for 16 points along x and y axis. BD_SENSOR_PROBE_NO_STOP is not enabled, however when enabled it errors out in a similar way. 

ABL worked fine with the BLtouch before, so I'm pretty sure the rest of the code is fine...

markniu, 2024-07-18:
which bugfix marlin do you use? you can try the version https://github.com/markniu/MarlinPULL/tree/bugfix-2.1.x 
there is a bug with BD_SENSOR_PROBE_NO_STOP if the ABL speed is fast in the official marlin github.

peterstevey, 2024-07-18:
That is the version that I'm using... I've disabled BD_SENSOR_PROBE_NO_STOP and still am getting the same error

markniu, 2024-07-19:
what's the error message?
how about disable the EEPROM and SDsupport for lite the firmware?

> Sorry - I have another issue now...
>
> Homing works well, and z height is accurate!
>
> When initiating G29, the axis are homed, but then marlin crashes and the printer restarts. Sometimes it beeps and [...]

closed 2025-02-13 by markniu