markniu/Bed_Distance_sensor #168: Query Only - bed_mesh 'fade' klipper config

KiwiChrist, 2024-07-29:
I have RTL setup to run on my modified Ender3v2 and is working great.

However I am seeing a slight variance in the first layer quality when doing a sparse infill across the bed as the Z obviously adjusts for the RTL.

What I'd like to know is...  
1. Does bed_mesh 'fade' if enabled effect RTL?
2. Can I increase the RTL max height to 0.4 or more?

Thanks.

markniu, 2024-07-29:
1. the bed_mesh should not effect the RTL if there is BED_MESH_CALIBRATE in the start gcode.
2. yes, you can set it with BDSENSOR_SET REAL_TIME_HEIGHT=0.41, the range of that height is 0~3mm.

KiwiChrist, 2024-07-29:
Thanks for the quick reply Mark,
I note that point 1 of RTL installation instruction states "This feature only works with the BDsensor and Z motors are connected in the same MCU".

Questions related to me running my Voron with a Huvud toolhead MCU now.

1. Why does it not work, if the MCU controlling the Z is not the same MCU with the BDsensor connected to it, curious?
2. Is this a feature that can be developed for those of us with BDsensors connected to toolhead MCU's? 

Thanks

markniu, 2024-07-30:
I had tried this with different mcu but failed. the reason is the structure of the klipper and the limit communication speed between different MCU. it can not read the data from the BDsensor in real time from one mcu to a non real time linux OS then send control command to mcu again while printing.

KiwiChrist, 2024-07-30:
Fair enough, makes sense.
Thanks.

I will need to think about my setup, maybe I have to add the long cable back to the toolhead to have the BDsensor connected to the main MCU again... I like the RTL it works flawlessly on my modified E3v2.