markniu/Bed_Distance_sensor #169: problem with M119

tjengbudi, 2024-08-07:
hello
i connected bd sensor to my skr 1.4
i using bltouch port from the board. but cause the probing is cannot works i connect it to exp1 port. for sure i connect it to port p1.27

the schematic is right here
https://user-images.githubusercontent.com/74151066/98479426-84961880-21f7-11eb-8f3c-1270e7dc9f71.png

my settimg is

[BDsensor] 
sda_pin: P1.27
scl_pin: P2.0
delay: 20
z_offset:0 
x_offset: 0
y_offset: 34
position_endstop: 0.5 
value is 0.5~1.0
collision_homing:0 
collision_calibrate:0

M119 always return z: open
when i added metal under the sensor the light is on. but the M119 still said z:Open

M102 s-1
BDsensorVer:V1.2c pandapi3d     ,switch_mode=1,collision_homing=0,collision_cal=0

m102 s-2
3.01mm

i checked cable with multitester it is not broken at all

installed newest bdsensor script and klipper last version.

so i do not understand why this happened

markniu, 2024-08-08:
did you calibrate the BDsensor? you can calibrate the sensor again.

tjengbudi, 2024-08-08:
> did you calibrate the BDsensor? you can calibrate the sensor again.

already calibrate, calling m102 s-5 lowest value is 55

calibrate again after you asked and after calibrate m102 s-5 lowest value is 53

and it is still not triggered the z endstop.

tjengbudi, 2024-08-08:
i trying to change sda pin with random pin in the board but i not connect it to bd sensor. i want to make sure that i might have wrong connection. or cable error.

after restart, command m102 s-1 replied "No data or corrupt data from BDsensor(), Please check connection"
so i am sure that i have connect it in the right pin and working

change it back and it replied "BDsensorVer:V1.2c pandapi3d     ,switch_mode=1,collision_homing=0,collision_cal=0"

must i flash it?

tjengbudi, 2024-08-08:
i use screw for the test looks like i do it wrong. test with wider dimension it triggered..

thankyou

closed 2024-08-08 by tjengbudi