markniu/Bed_Distance_sensor #170: collision_homing returning wrong value

soleshoe, 2024-08-10:
After setting 
collision_homing:1 
collision_calibrate:1
in printer.cfg, the z-offset returned by BDSENSOR_CALIBRATE is not accurate.
If i run a M102 S-6 after the G28 homing, the BDSENSOR_CALIBRATE does return accurate result.

in conclusion it seems that the colision_calibrate function is not returning accurate result.
Complete sequence run :

11:05 AM M84
11:05 AM 0.34mm
11:05 AM BDSENSOR_DISTANCE
11:05 AM G1 Z0.2
11:04 AM Collision: 1.1800 mm, Bed: 106.1C
11:04 AM G28
11:04 AM 0.21mm
11:04 AM BDSENSOR_DISTANCE
11:04 AM G1 Z0.2
11:04 AM Calibrate Finished!
11:04 AM 896
11:04 AM 891
11:04 AM 887
11:04 AM 882
11:04 AM 877
11:04 AM 871
11:04 AM 865
11:04 AM 858
11:04 AM 851
11:04 AM 843
11:04 AM 833
11:04 AM 823
11:04 AM 816
11:04 AM 806
11:04 AM 795
11:04 AM 779
11:04 AM 768
11:04 AM 757
11:04 AM 743
11:04 AM 727
11:04 AM 715
11:04 AM 698
11:04 AM 681
11:04 AM 664
11:04 AM 643
11:04 AM 624
11:04 AM 602
11:04 AM 578
11:04 AM 550
11:04 AM 526
11:04 AM 508
11:04 AM 485
11:04 AM 454
11:04 AM 422
11:04 AM 392
11:04 AM 370
11:04 AM 340
11:04 AM 311
11:04 AM 290
11:04 AM 276
11:03 AM Please Wait...
11:03 AM Calibrating, don't power off the printer
11:03 AM Collision: 1.1000 mm, Bed: 106.8C
11:03 AM Homing
11:03 AM M102 S-6
11:03 AM G1 Z0
11:02 AM G1 Z10
11:02 AM 0.31mm
11:02 AM BDSENSOR_DISTANCE
11:02 AM G1 Z0.2
11:00 AM SET_HEATER_TEMPERATURE HEATER=heater_bed TARGET=105
11:00 AM G1 Z10
10:59 AM 0.31mm
10:59 AM BDSENSOR_DISTANCE
10:59 AM G1 Z0.2
10:59 AM Collision: 1.1400 mm, Bed: 51.3C
10:59 AM G28

soleshoe, 2024-08-10:
heightmap calculation after G28 confirm this (nozzle & bed clean & cold)

[image]

soleshoe, 2024-08-10:
he problem may be related to colision_homing.
i have set colision_calibrate to 0 and here is what i get after a fresh calibration :
2:41 PM 0.59mm
2:41 PM BDSENSOR_DISTANCE
2:41 PM G1 Z0.2
2:41 PM Collision: 1.2080 mm, Bed: 30.1C
2:40 PM G28

soleshoe, 2024-08-10:
if i make the same test whithout G28 (so without calling collision_homing), here is what i get after a fresh calibration :
2:46 PM 0.01mm
2:46 PM BDSENSOR_DISTANCE
2:46 PM G1 Z0
2:46 PM 1.0mm
2:46 PM BDSENSOR_DISTANCE
2:45 PM G1 Z1
2:45 PM 0.21mm
2:45 PM BDSENSOR_DISTANCE
2:45 PM G1 Z0.2
2:45 PM Calibrate Finished!
2:45 PM 880
2:45 PM 876
2:45 PM 871
2:45 PM 859
2:45 PM 859
2:45 PM 853
2:45 PM 845
2:45 PM 837
2:45 PM 830
2:45 PM 823
2:45 PM 814
2:45 PM 806
2:45 PM 796
2:45 PM 785
2:45 PM 772
2:45 PM 760
2:45 PM 749
2:45 PM 737
2:45 PM 721
2:45 PM 708
2:45 PM 691
2:45 PM 676
2:45 PM 658
2:45 PM 641
2:45 PM 614
2:45 PM 604
2:45 PM 579
2:45 PM 561
2:45 PM 541
2:45 PM 513
2:45 PM 492
2:45 PM 442
2:45 PM 443
2:45 PM 416
2:45 PM 390
2:45 PM 364
2:45 PM 335
2:45 PM 309
2:45 PM 286
2:45 PM 270
2:45 PM Please Wait...
2:45 PM Calibrating, don't power off the printer
2:45 PM M102 S-6
2:45 PM SET_KINEMATIC_POSITION Z=200

markniu, 2024-08-13:
we are discussing it in the messenger of facebook, and will update the message here later.