markniu/Bed_Distance_sensor #173: Internal error on command:"Z_TILT_ADJUST"

NexGen-3D-Printing, 2024-08-16:
Hi,

I just did a full update of everything, Klipper, and both of my MCU's and now using the same config as before, I get this error.

I really need to stop doing updates, everytime, it breaks something.

Current Config:

[BDsensor] 
scl_pin: tool_board:gpio20
sda_pin: tool_board:gpio11
delay: 20
z_offset: 0
z_adjust: 0
x_offset: 0
y_offset: -53
no_stop_probe: true
speed: 0.8
position_endstop: 0.5
#collision_homing: 1 #  set it 1 to enable homing with nozzle collision sensing.
#collision_calibrate: 1 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.

Traceback (most recent call last):
  File "/home/pi/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/pi/klipper/klippy/webhooks.py", line 436, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/pi/klipper/klippy/gcode.py", line 229, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 211, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 137, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 189, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/pi/klipper/klippy/gcode.py", line 226, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 211, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 137, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/gcode.py", line 137, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/z_tilt.py", line 145, in cmd_Z_TILT_ADJUST
    self.probe_helper.start_probe(gcmd)
  File "/home/pi/klipper/klippy/extras/probe.py", line 439, in start_probe
    done = self._move_next()
  File "/home/pi/klipper/klippy/extras/probe.py", line 405, in _move_next
    res = self.finalize_callback(self.probe_offsets, self.results)
  File "/home/pi/klipper/klippy/extras/z_tilt.py", line 161, in probe_finalize
    new_params = mathutil.coordinate_descent(
  File "/home/pi/klipper/klippy/mathutil.py", line 20, in coordinate_descent
    best_err = error_func(params)
  File "/home/pi/klipper/klippy/extras/z_tilt.py", line 159, in errorfunc
    total_error += adjusted_height(pos, params)**2
  File "/home/pi/klipper/klippy/extras/z_tilt.py", line 153, in adjusted_height
    x, y, z = pos
TypeError: cannot unpack non-iterable NoneType object

markniu, 2024-08-16:
seems the klipper was not update success.   
[image]

NexGen-3D-Printing, 2024-08-16:
Your spot on, its an issue with MainsailOS, I could not get it to update to the latest release, I went and installed RPI OS Lite and used Kiuah to install everything, and all is working as it should be now, I think I will now block this machine from having internet access as I want it frozen in time :) the old saying, if it aint broke, dont fix it!