markniu/Bed_Distance_sensor #174: BD sensor on IDEX printer

Sundancer78, 2024-08-16:
Hello and good day, 
I would like to use BDSensor with an IDEX printer. The printer has 2 independent Z-axes with their own end stop switch. Do I need a BD sensor for each extruder or is one on extruder 1 enough? Do other parameters need to be set in the [BDSensor] section in the printer.cfg file? Or, in my case, for [stepper_z] or [stepper_z]. I couldn't find any information about BD sensors and IDEX printers, so I'm asking my questions here. Thank you in advance for your help.

markniu, 2024-08-16:
interesting, I also want to add it into the idex printer. I think the best way is to add to a bdsensor on each of the extruder. one for homing,mesh bed level and another one works as a distance sensor. but there need to add some marco and python code for the klipper.

Sundancer78, 2024-08-21:
Will you provide code in the near future?