markniu/Bed_Distance_sensor #175: [FR] implement different probing/homing speed for contact probing

Life0fBrian, 2024-08-21:
Is it possible to have a dedicated probing or homing speed parameter available for contact probing?
The `speed` parameter in the `[BDsensor]` section is for QGL or PROBE_ACCURACY only.
So you can lets say define `homing_speed: 30` under the z stepper section in Klipper and perform a fast contactless Z homing before QGL and afterwards perform a final contact Z homing with way smaller speeds like 2 or 3 mm/s.

Like your Z_Tilt/QGL macro this will speed up the whole process before starting an actual print and can take advantage of both the contactless and contacted probing possibility of the BD_Sensor.

Other eddy current probes offer such features and it would be just nice to have here too!

markniu, 2024-08-22:
yes, it is possible, need to rewrite the macro of G28, I will try it first and then update here.

Life0fBrian, 2024-08-22:
That would be great!
Thanks Mark!

markniu, 2024-08-23:
I have just tested that and updated the BDsensor.py and macro for this. 
wiki: https://github.com/markniu/Bed_Distance_sensor/wiki/Homing-with-contactless-and-contact-probing-together

Life0fBrian, 2024-08-23:
As I'm on holidays now I can't test it right now. 
But what I want is that when i set probing to contactless and I home Z that it uses a higher speed and when I set it to contact probing it uses the lower speed.
It should not perform two subsequent homings though that will save time too.

FWIW:
- home Z fast contactless
- QGL with your macro
- final contact homing of Z

So depending on the probe mode it should use the fast or the slow speed.