markniu/Bed_Distance_sensor #177: Z does not follow the mesh (motors stopped)

Liquidus-tech, 2024-08-24:
Good afternoon, I have this problem that I can't figure out: the z motors don't follow the created mesh, they are practically stationary...
Can you help me solve this? Thank you
[klippy.log](https://github.com/user-attachments/files/16736987/klippy.log)

luco85, 2024-08-25:
with klipper?

Liquidus-tech, 2024-08-25:
Yes

luco85, 2024-08-26:
If you make a bed mesh on every print start is strange, otherwise  check if in you start macro load the mesh created. Klipper do not load default mesh as in the past
[image]

dronecz, 2024-08-26:
Where do you have your sensor connected? Did you follow [this](https://github.com/markniu/Bed_Distance_sensor/wiki/How-to-run-Real-Time-Level)?

Liquidus-tech, 2024-08-28:
Hi, yes, I think I've done everything. In the meantime I updated to version 1.2c via Slink and Continue not following the mesh (I use Ideamaker). Home and QGL works fine. 
Launching commands 

G28
BDSENSOR_SET REAL_TIME_HEIGHT=0.3
G1Z0.2

I received the following error:

Unable to get response "I2CBDr".

 from .cgf I changed the delay from 20 to 11or 8 but nothing changed.

At this point I don't know how to proceed anymore...
[klippy.log](https://github.com/user-attachments/files/16788750/klippy.log)

markniu, 2024-08-29:
Hi, the real time level can not work with the mesh bed level, so you can remove the `BDSENSOR_SET REAL_TIME_HEIGHT` and use the normal mesh bed level first by using `BED_MESH_CALIBRATE  `

> Hi, yes, I think I've done everything. In the meantime I updated to version 1.2c via Slink and Continue not following the mesh (I use Ideamaker). Home and QGL works fine. Launching commands
>
> G28 [...]

closed 2025-03-11 by markniu