markniu/Bed_Distance_sensor #178: Bed Distance Sensor data error2:10.24

bombsquad85, 2024-08-29:
Need a little help here.

i have flashed the firmware several times and finally got it to install w/o throwing a warning about the firmware being dirty and needing to be updated.

now that that is out of the way i am getting issues when i check the connection and i get Bed Distance Sensor data error2:10.24.

i have the type M sensor and it is installed the same way as pictured in the github example. 

my mainboard is an Octopus 1.1
the BD config section is as follows:
[BDsensor] 
# Don't use aliases for the board pins
sda_pin: PE15 
scl_pin: PE14
delay: 20 # you can set it 10 if the BDsensor version is >=1.2
z_offset:0 #  within -0.6 to 0.6mm
x_offset: 0
y_offset: 20
no_stop_probe: # fast probe that the toolhead will not stop at the probe point,disable it by commenting out.
position_endstop: 1.2 #the triggered position, recommend value is 1~2.8
collision_homing:0 #  set it 1 to enable homing with nozzle collision sensing.
collision_calibrate:0 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.
#QGL_Tilt_Probe:0 #set 1 to enable probe up and down when do quad_gantry_level

the cable is securely connected to the exp1 module.
[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PE8, EXP1_2=PE7,
    EXP1_3=PE9, EXP1_4=PE10,
    EXP1_5=PE12, EXP1_6=PE13,    # Slot in the socket on this side
    EXP1_7=PE14, EXP1_8=PE15,
    EXP1_9=<GND>, EXP1_10=<5V>,

i thought everything was done correctly
https://github.com/markniu/Bed_Distance_sensor/wiki/BDsensor-M

any idea what i missed? i literally just spent over 2 hours trying to figure this out.

TIA

markniu, 2024-08-29:
Hello, what's hardware version on the BDsensorM,  there is a different of the SDA and SCL pin number between the V1.0 and V1.1,
which you can see from the text string on the pcb board.

bombsquad85, 2024-08-29:
> Hello, what's hardware version on the BDsensorM, there is a different of the SDA and SCL pin number between the V1.0 and V1.1, which you can see from the text string on the pcb board.

its a v1.1

bombsquad85, 2024-08-29:
> Hello, what's hardware version on the BDsensorM, there is a different of the SDA and SCL pin number between the V1.0 and V1.1, which you can see from the text string on the pcb board.

ok, i got it to work with the BL touch connectors. any reason why it didnt work with EXP1?

markniu, 2024-08-30:
the pin name should be configured as following for the BDsensorM v1.1 connecting to Octopus 1.1 with EXP1
```
sda_pin: PE10
scl_pin: PE9
```

bombsquad85, 2024-08-30:
> the pin name should be configured as following for the BDsensorM v1.1 connecting to Octopus 1.1 with EXP1
> ```
> sda_pin: PE10
> scl_pin: PE9
> ```
> 
> [image]
> 

Thank you for this response. I have since gotten it to work using the BL touch connectors and it is working wonderfully.