markniu/Bed_Distance_sensor #180: Triggering in mid air, only when bed is hot.

bombsquad85, 2024-09-07:
So i am loving this sensor, but after some testing and  using it i run into an issue where when i go to print abs and have the bed preheated to 100C the sensor triggers in mid air. i have tried 2nd homing speed of 4 and 5. is there another thing i am missing?

sensor is the type m by the way.

9:28 PM
warning: triggered in air, 737
9:28 PM
G28 Z
9:28 PM
Raw data:198 at 0 mm, BDsensor to bed: 0.7920 mm, Bed: 87.5C
9:28 PM
G28 Z
9:28 PM
Raw data:199 at 0 mm, BDsensor to bed: 0.7960 mm, Bed: 78.0C
9:27 PM
G28 Z
9:27 PM
Raw data:199 at 0 mm, BDsensor to bed: 0.7960 mm, Bed: 66.2C
9:27 PM
G28 Z
9:27 PM
Raw data:200 at 0 mm, BDsensor to bed: 0.8000 mm, Bed: 54.5C
9:27 PM
SET_HEATER_TEMPERATURE HEATER=heater_bed TARGET=100
9:27 PM
G28 Z

markniu, 2024-09-07:
did it home z axis success? normally, it will home z axis without problem but will take more time to touch the bed, also I am planning to reduce the time after that warning.

bombsquad85, 2024-09-07:
> did it home z axis success? normally, it will home z axis without problem but will take more time to touch the bed, also I am planning to reduce the time after that warning.

It does the first home, when it does the 2nd homing it throws the warning that it probed mid-air or something like that at 740. But it only does this when the printer is preheated