markniu/Bed_Distance_sensor #182: Sensor board flashes slowly and is always triggered when issuing a M119

PJawZK, 2024-09-22:
I got it installed, and got confirmation that it is connected by the M102 s-1 and M119.

But it is always triggered and always shows 0.0mm no matter if it is close to a metal surface or not.

When I first start the printer, the boards LED turns itself on and off, when I go near or away from a metal surface, indicating it works ok.
But when I issue for instance the M102 s-1 command, the LED starts to turn itself on and off slowly.
And if I put the BD Sensor close to a metal surface, it flashes on and off really fast.

markniu, 2024-09-23:
do you calibrated the sensor?  and also you can send gcode M102 S-7  to see what's the raw data returned from the sensor.

PJawZK, 2024-09-23:
Haven't been able to do that, cause of the constant "Z:TRIGGERED" issue.

But I will look into and see what that S-7 spits out in terms of information

PJawZK, 2024-09-23:
It reacts to changes from  going closer and closer to the material.

Mid-air value 952, going down incrementally as I move closer to the metal surface.
Lowest value 11, says it is at 0.0mm

When I put the probe directly on the surface, with no gap, I get a reading of 1015. It said it was at 10.15mm

And any other value in between, it will always stay at 0.0mm

But I read some other issues reported here, and you mentioned that is because it is not calibrated.

That still is not possible for me to do though, as I can't get past homing Z without Klipper shutting down on me

PJawZK, 2024-09-23:
I'm running a Dual Z-axis setup, 2 z-motors on different drivers

PJawZK, 2024-09-23:
So looked a bit more, the issue of not homing properly was due to my use of the [homing_override] and my badly implemented macro, that set the Z-axis as homed before it actually was homed.

Using the [safe_z_home] and deactivating the override solved this, but the board behaves the same.

Now that I can home without issue at least, I get this error from the BD sensor when issuing the m102 s-6;
warning: triggered in air, 747

markniu, 2024-09-23:
about the  triggered in air, you can try to increase the speed of the `second_homing_speed` in [steeper_z]

PJawZK, 2024-09-23:
I have, seemingly, gotten it to work with a lot of troubleshooting and changing settings, as well as going over wiring a bunch.

I used both the wire loom and the EXP1 connection, at the beginning none worked, now it seems it works with the wire loom approach, but not the EXP1 approach.
Got connection on both and coud recieve data from both, but it would not run a calibration with EXP1. But now does it wit the wrire loom.

I am uncertain what fixed it, and i have so far only made it home properly after finally being able to calibrate it.

So again, not sure what the root of the issue was here, but will look into it more closely and see if I can find a possible cause.

markniu, 2024-09-23:
btw, the order number of the pins maybe different, please see the actual text string on the sensor. https://github.com/markniu/Bed_Distance_sensor/issues/178#issuecomment-2319621934

PJawZK, 2024-09-23:
I will look into the pins more, and see if it was something I missed there

PJawZK, 2024-09-23:
The EXP1 port function works now as well. So no hardware issues.

Just bad config files it would seem, from all of my issues. From my end, not yours.
Unfortunately I can't document a proper fix for my issue, so I can't help out for anyone else experiencing issues like me.