markniu/Bed_Distance_sensor #183: M102 S-6 causes bed to drive into nozzle without stopping

tpups, 2024-10-02:
I have been following this guide to install bed distance sensor on an EBB36. The connection check succeeds. I move the bed so the nozzle is barely touching then send M102 S-6. The console says bd sensor is calibrating and not to turn off the printer. It slowly raises the bed, pushing the nozzle and hotend up further and further and flexing the linear rods. After several mm of movement I had to do an emergency stop.

What would cause this behavior?

```
# Motor3
[stepper_z]
step_pin: PB8
dir_pin: !PB7
enable_pin: !PE0
microsteps: 16
rotation_distance: 8
# endstop_pin: ^PF3
# position_endstop: 0
position_max: 230
position_min: -5.0
homing_speed: 8
second_homing_speed: 3
homing_retract_dist: 3

### added for BD Sensor
endstop_pin: probe:z_virtual_endstop
# position_endstop: 0.5
homing_speed: 5
second_homing_speed: 1  #set this to 3 if homing with collision
homing_retract_speed: 2
homing_retract_dist: 5

[safe_z_home]
home_xy_position: 100, 100
z_hop: 4.0
z_hop_speed: 5.0

[bed_mesh]
speed: 80
mesh_min: 60, 30
mesh_max: 230, 210
probe_count: 6, 6
horizontal_move_z: 0.8 # 0.7~1.0mm is recommended
algorithm: bicubic  # this should be bicubic if the bed mesh points count is > 6*6
zero_reference_position: 100, 100 # this should be the same as home_xy_position

[force_move] 
enable_force_move: true   # required by the command SET_KINEMATIC_POSITION in the calibration step below.

[tmc2209 stepper_z]
uart_pin: PB9
#diag_pin: PF2
run_current: 0.650
stealthchop_threshold: 999999
```

```
[BDsensor] 
# Don't use aliases for the board pins
sda_pin: EBBCan:PB8 
scl_pin: EBBCan:PB9
#scl_pin:MKS_THR:gpio20 # example of connecting to CAN module like MKS THR42 
#sda_pin:MKS_THR:gpio11 
#scl_pin:host:gpio17 # example of connecting to GPIO on RaspberryPi
#sda_pin:host:gpio27
delay: 20 # you can set it 10 if the BDsensor version is >=1.2
z_offset: 0 #  within -0.6 to 0.6mm
x_offset: -34
y_offset: 0
samples: 2
no_stop_probe: # fast probe that the toolhead will not stop at the probe point,disable it by commenting out.
position_endstop: 1.2 #the triggered position, recommend value is 1~2.8
collision_homing: 0 #  set it 1 to enable homing with nozzle collision sensing.
collision_calibrate: 0 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.

[force_move] 
enable_force_move: true   # required by the command SET_KINEMATIC_POSITION in the calibration step below.
```

tpups, 2024-10-02:
I have the BD sensor mounted 0.9mm above the nozzle.

With the nozzle ~20mm above the bed, M102 S-2 returns `BDsensor:Out of measure Range or too close to the bed`

With the nizzle <1mm above the bed, M102 S-2 returns `0.0mm`

So it seems like it can tell when it's close to the bed, but it hasn't been calibrated. And calibration seems intent on destroying my bed.

tpups, 2024-10-02:
I tried setting position_max: 230 for stepper_z in config then I used SET_KINEMATIC_POSITION Z=230 to tell the machine it was at max Z. Then I ran M102 S-6 again with the hope that the machine thinking it was at max Z it wouldn't let it push the nozzle into the bed.

It still did it.

tpups, 2024-10-02:
I have the BD sensor mounted 0.9mm above the nozzle.

With the nozzle 20mm above the bed, M102 S-2 returns `BDsensor:Out of measure Range or too close to the bed`

With the nozzle anywhere from ~0-3mm from the bed, M102 S-2 returns `0.0mm`

It can tell when it's close to the bed, but it hasn't been calibrated. Calibration seems to be able to push my bed above the kipper-defined max z and damage the bed.

markniu, 2024-10-02:
it seems the direction of the stepper_z is inverted. you can check that by 
SET_KINEMATIC_POSITION Z=10
G1 Z12
and to see if the direction of z axis is right?

tpups, 2024-10-02:
> it seems the direction of the stepper_z is inverted. you can check that by SET_KINEMATIC_POSITION Z=10 G1 Z12 and to see if the direction of z axis is right?

I think it's moving right - if I click +1 in klipper it moves up and -1 makes it move down.

tpups, 2024-10-02:
> it seems the direction of the stepper_z is inverted. you can check that by SET_KINEMATIC_POSITION Z=10 G1 Z12 and to see if the direction of z axis is right?

Yes that was the problem! Silly mistake on my part. Thank you!

closed 2024-10-02 by tpups