markniu/Bed_Distance_sensor #184: Poor first layer with bd sensor mesh

josh-hutchinson, 2024-10-07:
Hello, I have my bd sensor mostly set up and working. The only issue remaining is that the mesh it creates does not seem to be providing a good first layer. I am seeing good first layer on one side of the print, but on the other side there are gaps between the print lines. Looking at the generated mesh, the sensor has measured a low spot where the gaps are appearing, so it does see the unevenness in the bed there.

I wonder if the bd sensor isn't being accurate enough to get the nozzle offset correct? My first value returned is around 350 which is under the 400 as per the docs, although I see a first value around 100 is better. Would lowering the bd sensor likely improve my first layer, or is it more likely to be a different issue?

Thanks!

josh-hutchinson, 2024-10-07:
I was able to resolve this issue by adding adaptive_margin to my [bed_mesh] section.

closed 2024-10-07 by josh-hutchinson

tpups, 2024-10-09:
Well it turns out adding the margin didn't fully fix my issue. I'm back to considering lowering the bd sensor. Maybe it's accuracy isn't great if the first reading is around 350?

dhoard, 2025-02-26:
@tpups are using adaptive mesh? `BED_MESH_CALIBRATE ADAPTIVE=1`?

I updated yesterday to the latest Klipper + BD Sensor code, with no other changes, and suddenly started having bad first layers. Removing the use of adaptive mesh resolved the issue.

markniu, 2025-02-27:
Recommend to use the real_time_level instead of mesh bed leveling if you find it hard to get good first layer. but need to make sure the sensor will not go out of the bed plate while printing.