markniu/Bed_Distance_sensor #185: No data or corrupt data from BDsensor(), Please check connection

MDegelmann, 2024-10-07:
Hello, I can't get a connection
I use the Manta M8P and since the pins

sda_pin: PC9
scl_pin: PA8

Firmware is flashed and the version is: v0.12.0-316-g96cceed23-dirty

the sensor is v1.3

don't know what to do next

markniu, 2024-10-08:
how about use the probe pins:
sda_pin: PD13
scl_pin: PD12

MDegelmann, 2024-10-08:
The same problem No data or corrupt data from BDsensor( ), Please check connection

markniu, 2024-10-08:
does the light on while put a metal close the coil?  check again the 5v and other wire order.

MDegelmann, 2024-10-08:
111 / 5.000
Yes, the LED lights up when you hold a piece of metal to it.

I have checked all connections
GND
5V
SCL
GND
SDA

markniu, 2024-10-08:
What does the cable that comes with the BDsensor look like?

MDegelmann, 2024-10-08:
I measured the cable and everything has contact

markniu, 2024-10-08:
it can not be measured by multimeter sometimes.  what's the delay value in the BDsensor section? you can try to increase that value.

MDegelmann, 2024-10-08:
I was able to solve the problem.

I flashed the sensor and now it works

ceb0610, 2024-10-18:
Having the same problem but not having any luck flashing my sensor.

markniu, 2024-10-19:
> Having the same problem but not having any luck flashing my sensor.

which board and pins are used for the sensor?

ceb0610, 2024-10-19:
> > Having the same problem but not having any luck flashing my sensor.
> 
> which board and pins are used for the sensor?

BD Sensor V1.3
ST-Link V2 connected according to the wiki. I have checked the wiring multiple times.

markniu, 2024-10-19:
sorry, I mean which the pins are used for the sensor on the mainboard of the printer?

> > 
> > which board and pins are used for the sensor?
> 
> BD Sensor V1.3 ST-Link V2 connected according to the wiki. I have checked the wiring multiple times.
> 
> When I connect everything the sensor lights up when I get it close to metal, so I think the sensor is working.

ceb0610, 2024-10-19:
> sorry, I mean which the pins are used for the sensor on the mainboard of the printer?

Using the BLTouch interface on my toolhead board. Getting 5V from another connector.

[BDsensor] 
sda_pin:extra_mcu:PB6 
scl_pin:extra_mcu:PB7

[image]

markniu, 2024-10-19:
do you have the electronic schematic ? you can check if there is a capacitor on the PB6 on the board

ceb0610, 2024-10-19:
> do you have the electronic schematic ? you can check if there is a capacitor on the PB6 on the board

No capacitor.

MDegelmann, 2024-11-07:
Problem durch neues flashen der Hardware gelöst

closed 2024-11-07 by MDegelmann