markniu/Bed_Distance_sensor #190: Consist issues with triggering in midair

bombsquad85, 2024-10-25:
I'm running the type M. I think it's v1.1, but I'm getting consistent issues with the probe triggering in mid-air. I have dialed down the homing speeds, and it's only an issue when the bed is warmed up.

I also saw there is an update. Is there any info on what the update consists of?

markniu, 2024-10-25:
one of the reason is the homing speed is too slow,  you can increase up the homing speed a little.

bombsquad85, 2024-10-27:
> one of the reason is the homing speed is too slow, you can increase up the homing speed a little.

so i had to slow down the homing speed for it to not trigger in mid air for any homing action. is possible to calibrate this hot?

bombsquad85, 2024-11-14:
just wanted to update this, i recalibrated the sensor while the printer was hot and so far so good, i am running some more tests to make sure that everything is good, but i havent gotten the usually "triggered in midair" while the printer was heated

suroh1994, 2025-06-21:
I've done 6 consecutive prints, starting at the top left going right and then the second row left to right again. I wanted to dial in the z-offset so I adjusted it and noted the value on the pieces. Please ignore that I accidentally flipped the -0.1 text. 
[image]

Looking at the results, the first print (0.0) was too high and the second (-0.1) was still too high. However, the -0.2 result (print no. 3) was suddenly higher than the first print. I reran the print to check for errors and the same or even worse result. I dialed it back to -0.15 but the result was still poor and the final print was at -0.1 to check if it at least matched my second print but no, it was also too far off the build plate.

From my small testing, it appears to happen once the probe has heated up significantly. I haven't looked at the temperature in the chamber and I don't have anything to measure the probes temperature itself. What else can I try?