markniu/Bed_Distance_sensor #191: Always triggered

Nmorelm, 2024-10-25:
Trying to setup my BDsensorM v1.1 using EXP1 connection on BTT Octopus v1.1
Klipper shows probe as always triggered , no matter is it near bed or in mid-air. 
Thas my klipper config part for probe:
`[BDsensor] 
sda_pin: PE10
scl_pin: PE9
delay: 20 # you can set it 10 if the BDsensor version is >=1.2
z_offset:0 #  within -0.6 to 0.6mm
x_offset: 0
y_offset: -23
no_stop_probe: # fast probe that the toolhead will not stop at the probe point,disable it by commenting out.
position_endstop: 1.2 #the triggered position, recommend value is 1~2.8
collision_homing:0 #  set it 1 to enable homing with nozzle collision sensing.
collision_calibrate:0 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.
#QGL_Tilt_Probe:0 #set 1 to enable probe up and down when do quad_gantry_level`

```
08:25
x:open y:open z:TRIGGERED
08:25
m119
```

```
08:25
BDsensorVer:V1.2c pandapi3d     ,switch_mode=1,collision_homing=0,collision_cal=0
08:25
M102 S-1
```

```
08:27
0.0mm
08:27
M102 S-2
```

markniu, 2024-10-25:
it seems that sensor has not been calibrated.
the distance data is 0.0mm after M102 S-2 that is bellow the postion_endstop:1.2mm so it will output triggered.

Nmorelm, 2024-10-25:
I've tested sensor on custom printer. It gives same result - 0.0 mm mid-air , nozzle against the bed , so the sensor is 1.2 mm above the bed. Or even with metal ruler against the sensor. Still triggered and still 0.0 mm distance.

markniu, 2024-10-25:
did you calibrate the sensor?  how about the data after M102 S-5

Nmorelm, 2024-10-25:
Ok calibrated with M102 S-6:
```
18:38
Calibrate Finished!
18:38
823 at 3.9mm
18:38
820 at 3.8mm
18:38
815 at 3.7mm
18:38
812 at 3.6mm
18:38
807 at 3.5mm
18:38
804 at 3.4mm
18:38
797 at 3.3mm
18:38
793 at 3.2mm
18:38
787 at 3.1mm
18:38
780 at 3.0mm
18:38
774 at 2.9mm
18:38
766 at 2.8mm
18:38
759 at 2.7mm
18:38
753 at 2.6mm
18:38
744 at 2.5mm
18:38
737 at 2.4mm
18:38
727 at 2.3mm
18:38
717 at 2.2mm
18:38
706 at 2.1mm
18:38
696 at 2.0mm
18:38
686 at 1.9mm
18:38
672 at 1.8mm
18:38
660 at 1.7mm
18:38
647 at 1.6mm
18:38
632 at 1.5mm
18:38
619 at 1.4mm
18:38
602 at 1.3mm
18:38
586 at 1.2mm
18:38
568 at 1.1mm
18:38
552 at 1.0mm
18:38
532 at 0.9mm
18:38
512 at 0.8mm
18:38
491 at 0.7mm
18:38
471 at 0.6mm
18:38
447 at 0.5mm
18:38
425 at 0.4mm
18:38
401 at 0.3mm
18:38
378 at 0.2mm
18:38
354 at 0.1mm
18:38
333 at 0.0mm
18:38
Please Wait...
18:38
Calibrating, don't power off the printer
18:38
BDsensorVer:V1.2c pandapi3d     ,switch_mode=1,collision_homing=0,collision_cal=0
18:38
M102 S-6
```
data after M102 S-5:
```
18:39
823 at 3.9mm
18:39
820 at 3.8mm
18:39
815 at 3.7mm
18:39
812 at 3.6mm
18:39
807 at 3.5mm
18:39
804 at 3.4mm
18:39
797 at 3.3mm
18:39
793 at 3.2mm
18:39
787 at 3.1mm
18:39
780 at 3.0mm
18:39
774 at 2.9mm
18:39
766 at 2.8mm
18:39
759 at 2.7mm
18:39
753 at 2.6mm
18:39
744 at 2.5mm
18:39
737 at 2.4mm
18:39
727 at 2.3mm
18:39
717 at 2.2mm
18:39
706 at 2.1mm
18:39
696 at 2.0mm
18:39
686 at 1.9mm
18:39
672 at 1.8mm
18:39
660 at 1.7mm
18:39
647 at 1.6mm
18:39
632 at 1.5mm
18:39
619 at 1.4mm
18:39
602 at 1.3mm
18:39
586 at 1.2mm
18:39
568 at 1.1mm
18:39
552 at 1.0mm
18:39
532 at 0.9mm
18:39
512 at 0.8mm
18:39
491 at 0.7mm
18:39
471 at 0.6mm
18:39
447 at 0.5mm
18:39
425 at 0.4mm
18:39
401 at 0.3mm
18:39
378 at 0.2mm
18:39
354 at 0.1mm
18:39
333 at 0.0mm
18:39
M102 S-5
```

Nmorelm, 2024-10-25:
Yay! 
```
18:46
x:open y:open z:open
18:46
m119
```

markniu, 2024-10-26:
there should be more details in the klippy.log, or you can attach klippy.log file here

Nmorelm, 2024-10-26:
[klippy.log](https://github.com/user-attachments/files/17529602/klippy.log)
That my klippy.log

Nmorelm, 2024-10-26:
Also if i comment out `no_stop_probe:`
then klipper not crash at the bed mesh calibrate end, but gives an error:
`Bed Distance Sensor, out of range.:3.90`

closed 2024-10-27 by markniu