markniu/Bed_Distance_sensor #192: not really an issue, but a question

bombsquad85, 2024-10-28:
So I have been running my BD sensor in contact basically since install. i really like the autoz feature. i was just wondering is there a way to set it so that the z offset is slightly higher? i am finding that there is a little too much squish on my first layer and i have to babystep the z up.

Thanks in advance.

markniu, 2024-10-29:
yes, you still can adjust the z_offset as before.

bombsquad85, 2024-10-29:
> yes, you still can adjust the z_offset as before.

is there a link to this?

markniu, 2024-10-29:
like this: https://www.reddit.com/r/klippers/comments/oh0s50/how_to_save_z_adjust_after_tuning_during_print/
you can run SAVE_CONFIG  to save the z offset after babystep