markniu/Bed_Distance_sensor #195: BUG: warning: triggered in air, 1015

mninja79, 2024-11-09:
Of late I having been getting this bug when using contact probing from here: https://github.com/markniu/Bed_Distance_sensor/wiki/Homing-with-contactless-and-contact-probing-together. Also the same with just contact probing.
I have increase the  and decreased the homing speed but always same warning.
** NB: The major issue is that the bed continues to move in small increments. If I had not noticed this in time, may have ended up breaking something. If this happens during homing then it should error out not warn. **

Contactless does seem to work with no issues.

markniu, 2024-11-10:
do you adjust the homing speed with `second_homing_speed` ?
[Here](https://github.com/markniu/Bed_Distance_sensor/wiki/Homing-with-contactless-and-contact-probing-together) the speed of collision homing use the second_homing_speed,so need to increase the the second_homing_speed if there is warning: triggered in air

markniu, 2024-11-10:
one more test you can do is moving the nozzle close to the bed before collision homing e.g.`G1 Z1.2`

```
[gcode_macro G28]
rename_existing: G990028
gcode:
      {% if rawparams|length == 0 %}
           { action_respond_info("homing all") }
           G990028 X Y
           ## Homing with contactless probe with the homing_speed in [stepper_z]
           BDSENSOR_SET COLLISION_HOMING=0
           G990028 Z
           ## Homing with contact probe with the second_homing_speed in [stepper_z]
           BDSENSOR_SET COLLISION_HOMING=1
           G990028 Z
      {% else %}
            {% if 'x' in rawparams or 'X' in rawparams  %}
                 { action_respond_info("homing x") }
                 G990028 X
            {% endif %}
            {% if 'y' in rawparams or 'Y' in rawparams  %}
                 { action_respond_info("homing y") }
                 G990028 Y
            {% endif %}
            {% if 'z' in rawparams or 'Z' in rawparams  %}
                 { action_respond_info("homing z") }
                 ## Homing with contactless probe with the homing_speed in [stepper_z]
                 BDSENSOR_SET COLLISION_HOMING=0
                 G990028 Z
                 G1 Z1.2
                 ## Homing with contact probe with the second_homing_speed in [stepper_z]
                 BDSENSOR_SET COLLISION_HOMING=1
                 G990028 Z
            {% endif %}
      {% endif %}
```

mninja79, 2024-11-10:
Still the same issue with the tests above. I also tried to use just normal collision homing and it triggered when the bed started moving to home Z.

````
Endstop z still triggered after retract
warning: triggered in air, 1015
Internal error on command:"G28"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 211, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/safe_z_home.py", line 79, in cmd_G28
    self.prev_G28(g28_gcmd)
  File "/home/pi/klipper/klippy/extras/homing.py", line 281, in cmd_G28
    kin.home(homing_state)
  File "/home/pi/klipper/klippy/kinematics/corexy.py", line 64, in home
    homing_state.home_rails([rail], forcepos, homepos)
  File "/home/pi/klipper/klippy/extras/homing.py", line 217, in home_rails
    raise self.printer.command_error(
gcode.CommandError: Endstop z still triggered after retract

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/BDsensor.py", line 236, in _handle_command_error
    self.multi_probe_end()
  File "/home/pi/klipper/klippy/extras/BDsensor.py", line 247, in multi_probe_end
    self.mcu_probe.multi_probe_end()
  File "/home/pi/klipper/klippy/extras/BDsensor.py", line 1376, in multi_probe_end
    self.adjust_probe()
  File "/home/pi/klipper/klippy/extras/BDsensor.py", line 1353, in adjust_probe
    adj_z,adj_raw = self.adjust_probe_up(0.1,0,1)
  File "/home/pi/klipper/klippy/extras/BDsensor.py", line 1290, in adjust_probe_up
    self.toolhead.manual_move([None, None, homepos[2]], 2)
  File "/home/pi/klipper/klippy/toolhead.py", line 481, in manual_move
    self.move(curpos, speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 469, in move
    self.kin.check_move(move)
  File "/home/pi/klipper/klippy/kinematics/corexy.py", line 81, in check_move
    self._check_endstops(move)
  File "/home/pi/klipper/klippy/kinematics/corexy.py", line 74, in _check_endstops
    raise move.move_error("Must home axis first")
gcode.CommandError: Must home axis first: 150.000 150.000 1.300 [0.000]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 211, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 189, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/pi/klipper/klippy/gcode.py", line 226, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 214, in _process_commands
    self.printer.send_event("gcode:command_error")
  File "/home/pi/klipper/klippy/klippy.py", line 223, in send_event
    return [cb(*params) for cb in self.event_handlers.get(event, [])]
  File "/home/pi/klipper/klippy/klippy.py", line 223, in <listcomp>
    return [cb(*params) for cb in self.event_handlers.get(event, [])]
  File "/home/pi/klipper/klippy/extras/BDsensor.py", line 238, in _handle_command_error
    raise Exception("Multi-probe end")
Exception: Multi-probe end
Transition to shutdown state: Internal error on command:"G28"
Dumping gcode input 0 blocks

```

markniu, 2024-11-13:
sorry for not noticing your replay on time, 
what's the value of z_hop and homing_retract_dist? they should be >=5

https://github.com/markniu/Bed_Distance_sensor/issues/100#issuecomment-1976644529

> Still the same issue with the tests above. I also tried to use just normal collision homing and it triggered when the bed started moving to home Z.
> 
> ```

mninja79, 2024-11-13:
No worries.
It is set to 5

mninja79, 2024-11-13:
I have already tried that. same issue occurs.
Collision Calibration is not working now it worked yesterday. 
When I run command m102 s-6 it will home  x, y and when it starts to home Z it says endstop already triggered or retract.
I'm using a monster 8 V2 and sensor is connected to PB2 and PE10.
Contactless still works.

mninja79, 2024-11-17:
I went ahead and reflashed my board and upped the can bus rate to 1000000 now everything seems to work. no more warnings of 
triggered in air. Not sure what happened but all is good now. Weird stuff.
Thanks for the help in troubleshooting.

closed 2024-11-17 by mninja79

mninja79, 2024-11-21:
I found out what was causing the issue. There was interference between the  from the power cable running to toolhead mcu. The sensor cables were running parallel to it. Once I moved  them the issue stopped.  Need to get shielded cables. 
But for now I have switch to cartographer  only cause of the CAN option.  I'll move this sensor to my rook.   Keep up with the good work.