markniu/Bed_Distance_sensor #196: Crash On BED_MESH_CALIBRATE

leftouterjoins, 2024-11-09:
I have my sensor installed and it is working well so far, homing works and I've done some small prints.

However, when I try to calibrate the bed mesh I am getting a crash. The core error is:

```
stepcompress o=2 i=0 c=4 a=0: Invalid sequence
```
[klippy-error.log](https://github.com/user-attachments/files/17688548/klippy-error.log)

markniu, 2024-11-10:
The first error message in the log file is the actual error.
```
Unhandled exception during run
Traceback (most recent call last):
  File "/home/josh/klipper/klippy/klippy.py", line 176, in run
    self.reactor.run()
  File "/home/josh/klipper/klippy/reactor.py", line 292, in run
    g_next.switch()
  File "/home/josh/klipper/klippy/reactor.py", line 340, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/josh/klipper/klippy/reactor.py", line 158, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/josh/klipper/klippy/extras/BDsensor.py", line 336, in scan_sample_event
    intd = self.mcu_probe.BD_Sensor_Read(0)
  File "/home/josh/klipper/klippy/extras/BDsensor.py", line 857, in BD_Sensor_Read
    raise self.printer.command_error("Bed Distance Sensor, "
gcode.CommandError: Bed Distance Sensor, out of range.:10.15 
Transition to shutdown state: Unhandled exception during run
```

here the error is `Bed Distance Sensor, out of range`  while doing calibration. I think you can do z_tilt or QGL first

leftouterjoins, 2024-12-06:
Thanks, it doesn't really make any sense to me that the sensor is out of range though because right before attempting calibration I home everything (G28) and the log shows `Raw data:287 at 0 mm, BDsensor to bed: 1.1480 mm, Bed: 27.0C` so it clearly is within range. The error happens immediately when calibration begins and the nozzle doesn't move.

leftouterjoins, 2024-12-13:
I tinkered around with this some more. I edited the provided STL for the BL Touch mount to be 41mm. I re-calibrated everything and then it began getting past this error to the point where it would move to the first position in the scan. Then it would crash with a similar error. The problem there was the sensor was just hanging off the edge of the plate. Adjusting `mesh_min` and `mesh_max` settings fixed this and now I can get a height map!

closed 2024-12-13 by leftouterjoins