markniu/Bed_Distance_sensor #198: No trigger on probe after full movement

terri1769, 2024-11-29:
Switched from a Raspberry Pi to a NUC.  Since then I can't get Z tilt to work and sometimes even homing fails.

Procedure:
Followed calibration procedure per wiki.  I cannot get 10 between the first and second measurements no matter what I try.  Measurements currently from S102 S-5 are:
96 at 0.0mm
101 at 0.1mm
122 at 0.2mm
... etc.

Send G28 to home.  Sometimes it works, sometimes it gives the error in the title.
If it succeeded, click Z-Tilt.  It tries to do the Z tilt procedure and stops with "No trigger on probe after full movement".  
Click Z tilt again and it gives the same error followed by a shutdown with timer too close on MCU.

Klippy.log is attached.

[klippy.log](https://github.com/user-attachments/files/17960815/klippy.log)
[printer.txt](https://github.com/user-attachments/files/17960878/printer.txt)

Top -i shows the CPU is not getting bogged down and the only things running on the NUC are Klipper, Crowsnest, BDsensor, and one webcam.  Removing the webcam did not help the situation.

I've been through numerous calibrations and am using the gcode_macro Z_TILT_ADJUST as recommended.  Don't know what I'm missing.

terri1769, 2024-11-29:
Forgot to upload the config for the EBB36
[cancard.txt](https://github.com/user-attachments/files/17961232/cancard.txt)

markniu, 2024-11-30:
I suspend that the actual error should be the as the timer too close.  you can slow down the speed of homing and z tilt or adding G4 P1000 delay before G28 Z_TILT_ADJUST  in all the gcode_macro that was inspired by my sk1 printer https://github.com/tomsbasement/twotrees-sk1/blob/main/configurations/base-2.02/printer.cfg

markniu, 2024-12-07:
Recommend to decrease the `delay `value in `BDsensor `section from 20 to 10, because it can slow down the time consumption of klipper software

terri1769, 2024-12-07:
System quit booting so I'm having to rebuild from scratch and running into other problems.  I'll see if the problem still exists as soon as I can get it back to a running state

terri1769, 2024-12-08:
Problem appears to be resolved now.

closed 2024-12-08 by terri1769