markniu/Bed_Distance_sensor #199: warning:triggered at 0mm

GoTVm, 2024-12-02:
Just installed the BD Sensor on my new Klipper build. Whenever I do the single reading, I always get the error in the title; when I home the printer, it stops well above the bed (2-3mm) and then wont allow me to calibrate because it is too far from the bed:
`BDsensor is too far from the bed:934`

Printer.cfg:
```
[include fluidd.cfg]
[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f103xe_37FFD8054242353910582557-if00

[virtual_sdcard]
path: /home/dell/printer_data/gcodes
on_error_gcode: CANCEL_PRINT

[stepper_x]
step_pin: PB13
dir_pin: !PB12
enable_pin: !PB14
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC0
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_x]
uart_pin: PB15
run_current: 0.580
stealthchop_threshold: 999999

[stepper_y]
step_pin: PB10
dir_pin: !PB2
enable_pin: !PB11
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC1
position_endstop: 235
position_max: 235
homing_speed: 50
homing_positive_dir: true

[tmc2209 stepper_y]
uart_pin: PC6
run_current: 0.580
stealthchop_threshold: 999999

[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
#endstop_pin: ^PC2
#position_endstop: 0.0
position_max: 250
homing_speed: 5
second_homing_speed:3

[tmc2209 stepper_z]
uart_pin: PC10
run_current: 0.580
stealthchop_threshold: 999999

[extruder]
step_pin: PB3
dir_pin: !PB4
enable_pin: !PD2
microsteps: 16
rotation_distance: 33.500
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC8
sensor_type: ATC Semitec 104NT-4-R025H42G
sensor_pin: PA0
control: pid
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250

[tmc2209 extruder]
uart_pin: PC11
run_current: 0.650
stealthchop_threshold: 999999

[heater_bed]
heater_pin: PC9
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC3
control: pid
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

#[fan]
#pin: PA8

[safe_z_home]
home_xy_position: 115,115

[bed_mesh]
speed: 120
mesh_min: 20,20
mesh_max: 220,220
probe_count: 6,6
horizontal_move_z: 1
algorithm: bicubic

[BDsensor] 
sda_pin: PC14 # example of connecting to main board Creality V4.2.7
scl_pin: PA1
delay: 20 # you can set it 10 if the BDsensor version is >=1.2
z_offset:0 #  within -0.6 to 0.6mm
z_adjust: 0.0
x_offset: 0
y_offset: 49
#no_stop_probe: # fast probe that the toolhead will not stop at the probe point,disable it by commenting out.
speed: 0.8
position_endstop: 0.8 #the triggered position, recommend value is 1~2.8
collision_homing:0 #  set it 1 to enable homing with nozzle collision sensing.
collision_calibrate:0 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.
#QGL_Tilt_Probe:0 #set 1 to enable probe up and down when do quad_gantry_level

[printer]
kinematics: corexy
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[force_move]
enable_force_move: true

[static_digital_output usb_pullup_enable]
pins: !PC13

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PB6, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB7, EXP1_10=<5V>

# See the sample-lcd.cfg file for definitions of common LCD displays.
```

M102 S-1 output:
`// BDsensorVer:V1.2c pandapi3d     ,switch_mode=1,collision_homing=0,collision_cal=0`

markniu, 2024-12-03:
please move the nozzle just touch the bed and then send M102 S-6 to calibrate or set the collision_homing:1 and collision_calibrate:1 to let it auto homing before calibrate

GoTVm, 2024-12-03:
What do collision homing and collision calibrate do? I disabled them because the tutorial I was following didn't use them and because they are not very clearly documented

markniu, 2024-12-03:
it will homing itself and move the nozzle to the zero position and then calibrate itself if the collision_calibrate:1 after M102 S-6 .

about the collision : https://github.com/markniu/Bed_Distance_sensor/wiki/Collision-sensing

GoTVm, 2024-12-03:
So basically I can either:
- move the nozzle close to the bed by hand and then calibrate
- activate collision sensing and then just auto homing

correct?

About collision sensing, I read the documentation and have a couple questions:

The sensitivity depends on the speed, the slower speed the higher sensitivity, so you need to adjust the speed of Z axis, the homing_speed and second_homing_speed should be in the range of 2 ~ 5.

Should I try each value between 2 and 5 or is there a value that I can start from? Also, should homing speed and second homing speed be the same?

Won't the bed get scratched if the nozzle touches every time in the same spot to find the zero position?

markniu, 2024-12-03:
correct, 
you can use this value:

```
homing_speed: 5
second_homing_speed:3
homing_retract_dist:5
homing_retract_speed:2
```
like in this video: https://www.youtube.com/watch?v=RuPoXbrSPDc&feature=youtu.be

GoTVm, 2024-12-03:
Okay I will follow the video and update the post

GoTVm, 2024-12-03:
I activated collision sensing but now it says triggered in air, 748 and the z axis keeps moving up until it says move out of range 115 115 -0.1

GoTVm, 2024-12-03:
Managed to get the calibration done without collision sensing, now it reads always 0.1mm higher

GoTVm, 2024-12-03:
Also, when I move to 0mm, it reads higher values between 0.4 and 0.6mm, which isn't great for leveling precision as its a whole 2-3 layers high.

GoTVm, 2024-12-03:
Cannot get collision sensing to work, down here is the console output and my config
```
22:26:24  $ M102 S-6
22:26:25  // BDsensorVer:V1.2c pandapi3d     ,switch_mode=1,collision_homing=1,collision_cal=1
22:26:25  // Homing
22:27:34  !! Move out of range: 115.000 115.000 -0.100 [0.000]
22:27:34  !! Move out of range: 115.000 115.000 -0.100 [0.000]
```

Printer.cfg
```
[include fluidd.cfg]
[include shell_command.cfg]

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f103xe_37FFD8054242353910582557-if00

[virtual_sdcard]
path: /home/dell/printer_data/gcodes
on_error_gcode: CANCEL_PRINT

[stepper_x]
step_pin: PB13
dir_pin: !PB12
enable_pin: !PB14
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC0
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_x]
uart_pin: PB15
run_current: 0.580
stealthchop_threshold: 999999

[stepper_y]
step_pin: PB10
dir_pin: !PB2
enable_pin: !PB11
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC1
position_endstop: 235
position_max: 235
homing_speed: 50
homing_positive_dir: true

[tmc2209 stepper_y]
uart_pin: PC6
run_current: 0.580
stealthchop_threshold: 999999

[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
#endstop_pin: ^PC2
#position_endstop: 0.0
position_max: 250
homing_speed: 5
second_homing_speed:3
homing_retract_dist:5
homing_retract_speed:2

[tmc2209 stepper_z]
uart_pin: PC10
run_current: 0.580
stealthchop_threshold: 999999

[extruder]
step_pin: PB3
dir_pin: !PB4
enable_pin: !PD2
microsteps: 16
rotation_distance: 33.500
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC8
sensor_type: ATC Semitec 104NT-4-R025H42G
sensor_pin: PA0
control: pid
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250

[tmc2209 extruder]
uart_pin: PC11
run_current: 0.650
stealthchop_threshold: 999999

[heater_bed]
heater_pin: PC9
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC3
control: pid
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

#[fan]
#pin: PA8

[safe_z_home]
home_xy_position: 115,115
z_hop: 4.0
z_hop_speed: 10.0

[bed_mesh]
speed: 120
mesh_min: 10,30
mesh_max: 225,200
probe_count: 6,6
horizontal_move_z: 1
algorithm: bicubic
zero_reference_position:115,115

[BDsensor] 
sda_pin: PC14
scl_pin: PC12#PA1
delay: 20 # you can set it 10 if the BDsensor version is >=1.2
z_offset:0 #  within -0.6 to 0.6mm
z_adjust: 0.0
x_offset: 0
y_offset: 22
#no_stop_probe: # fast probe that the toolhead will not stop at the probe point,disable it by commenting out.
#speed: 0.8
position_endstop: 0.7 #the triggered position, recommend value is 1~2.8
collision_homing:1 #  set it 1 to enable homing with nozzle collision sensing.
collision_calibrate:1 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.
#QGL_Tilt_Probe:0 #set 1 to enable probe up and down when do quad_gantry_level

[printer]
kinematics: corexy
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[force_move]
enable_force_move: true

[static_digital_output usb_pullup_enable]
pins: !PC13

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PB6, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB7, EXP1_10=<5V>

# See the sample-lcd.cfg file for definitions of common LCD displays.
```

GoTVm, 2024-12-04:
after hours of testing, still havent gotten anywhere with the sensor. After trying to set up collision homing, it thinks its mounted at 4.06mm (even though it is mounted at 1.3mm), I can't get through the calibration process because:
- It will home, and spew a random value 
`// Raw data:1015 at 0 mm, BDsensor to bed: 4.0600 mm, Bed: 20.5C`
- I'll move the bed to touch the nozzle with SET_KINEMATICS_POSITION
- I'll send M102 S-6, it will home again then read this
`Raw data:1015 at 0 mm, BDsensor to bed: 4.0600 mm, Bed: 19.7C`
then it'll just say it can't calibrate because it is too far from the bed

I had it working for a little while, as you can see in the previous comments, now it's just a mess of random values.

Updated printer.cfg
```
[include fluidd.cfg]
[include shell_command.cfg]

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f103xe_37FFD8054242353910582557-if00

[virtual_sdcard]
path: /home/dell/printer_data/gcodes
on_error_gcode: CANCEL_PRINT

[stepper_x]
step_pin: PB13
dir_pin: !PB12
enable_pin: !PB14
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC0
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_x]
uart_pin: PB15
run_current: 0.580
stealthchop_threshold: 999999

[stepper_y]
step_pin: PB10
dir_pin: !PB2
enable_pin: !PB11
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC1
position_endstop: 235
position_max: 235
homing_speed: 50
homing_positive_dir: true

[tmc2209 stepper_y]
uart_pin: PC6
run_current: 0.580
stealthchop_threshold: 999999

[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
#endstop_pin: ^PC2
#position_endstop: 0.0
position_min: -6
position_max: 250
homing_speed: 4
second_homing_speed:3
homing_retract_dist:5
homing_retract_speed:1

[tmc2209 stepper_z]
uart_pin: PC10
run_current: 0.580
stealthchop_threshold: 999999

[extruder]
step_pin: PB3
dir_pin: !PB4
enable_pin: !PD2
microsteps: 16
rotation_distance: 33.500
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC8
sensor_type: ATC Semitec 104NT-4-R025H42G
sensor_pin: PA0
control: pid
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250

[tmc2209 extruder]
uart_pin: PC11
run_current: 0.650
stealthchop_threshold: 999999

[heater_bed]
heater_pin: PC9
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC3
control: pid
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

#[fan]
#pin: PA8

[safe_z_home]
home_xy_position: 115,115
z_hop: 4.0
z_hop_speed: 1

[bed_mesh]
speed: 120
mesh_min: 10,30
mesh_max: 225,200
probe_count: 6,6
horizontal_move_z: 1
algorithm: bicubic
zero_reference_position:115,115

[BDsensor] 
sda_pin: PC14
scl_pin: PC12#PA1
delay: 20 # you can set it 10 if the BDsensor version is >=1.2
z_offset:0 #  within -0.6 to 0.6mm
z_adjust: 0.0
x_offset: 0
y_offset: 22
#no_stop_probe: # fast probe that the toolhead will not stop at the probe point,disable it by commenting out.
#speed: 0.8
position_endstop: 1.2 #the triggered position, recommend value is 1~2.8
collision_homing:1 #  set it 1 to enable homing with nozzle collision sensing.
collision_calibrate:1 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.
#QGL_Tilt_Probe:0 #set 1 to enable probe up and down when do quad_gantry_level

[printer]
kinematics: corexy
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[force_move]
enable_force_move: true

[static_digital_output usb_pullup_enable]
pins: !PC13

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PB6, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB7, EXP1_10=<5V>

[bed_screws]
screw1: 30.5, 37
screw2: 30.5, 207
screw3: 204.5, 207
screw4: 204.5, 37

# See the sample-lcd.cfg file for definitions of common LCD displays.

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	  -0.972738, -0.772738, -0.552738, -0.442738, -0.182738, 0.137262
#*# 	  -0.712738, -0.572738, -0.372738, -0.202738, 0.127262, 0.567262
#*# 	  -0.552738, -0.432738, -0.182738, 0.027262, 0.407262, 0.897262
#*# 	  -0.472738, -0.322738, -0.022738, 0.257262, 0.687262, 1.097262
#*# 	  -0.472738, -0.232738, 0.127262, 0.467262, 0.907262, 1.097262
#*# 	  -0.502738, -0.172738, 0.257262, 0.627262, 1.097262, 1.097262
#*# x_count = 6
#*# y_count = 6
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = bicubic
#*# tension = 0.2
#*# min_x = 10.0
#*# max_x = 225.0
#*# min_y = 30.0
#*# max_y = 200.0

```

markniu, 2024-12-04:
the reason is that the nozzle was still touch on the bed after it move 0.6mm while calibrating, 
seems you let the nozzle too close to the bed before sending calibration command.

> Also, when I move to 0mm, it reads higher values between 0.4 and 0.6mm, which isn't great for leveling precision as its a whole 2-3 layers high.
> 
> EDIT: my calibration values do not match the requirements you specify in the wiki:
> 
> ```
> 21:56:38  // 237 at 0.0mm
> 21:56:38  // 237 at 0.1mm
> 21:56:38  // 239 at 0.2mm
> 21:56:38  // 241 at 0.3mm
> 21:56:38  // 243 at 0.4mm
> 21:56:38  // 246 at 0.5mm
> 21:56:38  // 251 at 0.6mm
> 21:56:38  // 265 at 0.7mm
> 21:56:38  // 279 at 0.8mm
> 21:56:38  // 294 at 0.9mm
> 21:56:38  // 309 at 1.0mm
> 21:56:38  // 325 at 1.1mm
> 21:56:38  // 341 at 1.2mm
> 21:56:38  // 357 at 1.3mm
> 21:56:38  // 364 at 1.4mm
> 21:56:38  // 388 at 1.5mm
> 21:56:38  // 405 at 1.6mm
> 21:56:38  // 421 at 1.7mm
> 21:56:38  // 436 at 1.8mm
> 21:56:38  // 451 at 1.9mm
> 21:56:38  // 466 at 2.0mm
> 21:56:38  // 481 at 2.1mm
> 21:56:38  // 495 at 2.2mm
> 21:56:38  // 508 at 2.3mm
> 21:56:38  // 522 at 2.4mm
> 21:56:38  // 537 at 2.5mm
> 21:56:38  // 551 at 2.6mm
> 21:56:38  // 565 at 2.7mm
> 21:56:38  // 579 at 2.8mm
> 21:56:38  // 593 at 2.9mm

GoTVm, 2024-12-04:
That could explain the imprecision (which isn't true, the nozzle wasn't jammed into the bed but rather gently positioned), but doesn't explain why I get 1015 and a bdsensor -bed distance of 4mm with collision sensing.

markniu, 2024-12-04:
one more test you can do,  read the raw data at different position directly by M102 S-7, that can help to understand how it works

[image]

GoTVm, 2024-12-04:
I understand generally how the device in itself works, and I manage to get it to work without collision sensing. I'd like to use collision sensing as well, but it doesn't seem to work as straightforward as without. As I said, it touches the bed during homing then reads 4mm height coil to bed, which is impossible.

image.png (view on web)<https://github.com/user-attachments/assets/484bd752-7bcd-42f5-b144-4871ab70a7d2>

GoTVm, 2024-12-04:
I tried reverting back to the configuration you shared on github, the probe is mounted at around 1mm from the nozzle. Every time I home, it sees a different sensor height:
Homed once and got
- Raw data 351 at 0mm, BDSensor to bed: 1.4mm,...

then I homed again and got
- Raw data 640 at 0mm, BDsensor to bed 2.56mm,...

bombsquad85, 2024-12-04:
> one more test you can do, read the raw data at different position directly by M102 S-7, that can help to understand how it works [...]

Hey Mark. I've been working with GoTVm on this and it is an interesting case. He has shown me pictures of the set up and we have gone through the tests extensively. When he goes through the proper steps things are looking alright until the results come back and he is getting this odd 1015 at 0mm. Do you think that there could be a miscommunication in the set up? I'm wondering if reinstalling everything is worth a shot.

markniu, 2024-12-05:
This is strange. especially the Raw data 640 at 0mm , 
how about set the second homing speed from 3mm/s to 4mm/s? 

> I tried reverting back to the configuration you shared on github, the probe is mounted at around 1mm from the nozzle. Every time I home, it sees a different sensor height: Homed once and got - Raw [...]

markniu, 2024-12-05:
what's the data return by M102 S-7 after " 1015 at 0mm" ?
yes, may be there is a miscommunication , for test you can also increase the delay value from 20 to 30 in the section [BDsensor].

> Hey Mark. I've been working with GoTVm on this and it is an interesting case. He has shown me pictures of the set up and we have gone through the tests extensively. When he goes through the proper [...]

GoTVm, 2024-12-05:
> This is strange. especially the Raw data 640 at 0mm , how about set the second homing speed from 3mm/s to 4mm/s?

I tried that on and off, played with speeds a lot but that never seemed to make a difference. I then went to watch your videos on YouTube and noticed that your bed was moving much faster than mine, which prompted a check on my printer's kinematics. 

That's when I found out that I was missing the gear_ratio parameter in my stepper_z configuration, causing my bed to move half as much as it should've (and generally in an erratic manner).

Since fixing the gear ratio, I've been able to get pretty consistent results from homing multiple times in a row (the sensor reads a distance of 0.3mm from the bed, +-0.05), though the accuracy is much lower than the one you advertise (+-0.005, one order of magnitude more precise).

GoTVm, 2024-12-05:
> what's the data return by M102 S-7 after " 1015 at 0mm" ? yes, may be there is a miscommunication , for test you can also increase the delay value from 20 to 30 in the section [BDsensor].

I didn't have the chance to get that value when it was still showing 1015 at 0mm, if it ever happens again I'll be posting it here.

Now, since I got it working, I'd like to make it more repeatable and have accuracy closer to the one you advertise. How would I do that?

markniu, 2024-12-06:
is the 0.3mm returned after G28? like `at 0 mm, BDsensor to bed: 0.3 mm` ,if it is that is not the max difference, but the mount distance of coil.

> Since fixing the gear ratio, I've been able to get pretty consistent results from homing multiple times in a row (the sensor reads a distance of 0.3mm from the bed, +-0.05), though the accuracy is much lower than the one you advertise (+-0.005, one order of magnitude more precise).

markniu, 2024-12-07:
Recommend to decrease the delay value in BDsensor section from 20 to 10, because it can slow down the time consumption of klipper software

GoTVm, 2024-12-07:
What do you mean by slowing down the time consumption of klipper? Just during the probing process?

I'll decrease it when I get home.