markniu/Bed_Distance_sensor #202: Internal error on command:"BED_MESH_CALIBRATE"

04srt10, 2024-12-19:
while attempting to perform a heightmap,

6:16 PM
Internal error on command:"BED_MESH_CALIBRATE"
6:16 PM
Klipper state: Shutdown
6:16 PM
probe at 10.000,190.000 is z=0.000000

also, with  NO_STOP_PROBE enabled, it still stops at every point. 

This is on a E3NG with an SKR MINI E3 V2  with a bd sensor.

04srt10, 2024-12-20:
[klippy.log](https://github.com/user-attachments/files/18205235/klippy.log)

04srt10, 2024-12-20:
We resolved the issue with a manual klipper update! please close my ticket! thank you Mark

harlequincirrus, 2025-01-12:
Hi there, i still have the same issue. I've updated all components, reinstalled klipper and the BD Sensor. But nothing seems to Work.
I'm as well using an e3ng with an skr 3 ez.

markniu, 2025-01-13:
What's the first error message in the klippy.log?

the klipper should be update to the latest too if the error is same as o4srt10's klippy.log :

[image]

> Hi there, i still have the same issue. I've updated all components, reinstalled klipper and the BD Sensor. But nothing seems to Work. I'm as well using an e3ng with an skr 3 ez.

closed 2025-03-11 by markniu