markniu/Bed_Distance_sensor #204: Wire plug in Connection1.jpg Wrong

brosskgm, 2024-12-29:
Just received your unit Today. Installing on Mega X with SKR V1.3 TMC 2208

Your Connection1.jpg picture in your instructions is not what is actually pluged in. Does not have the same colors as BLTouch install(BLTOUCHWIRESETUP Attached).

Also configuration setup in pins file mentions a SCL_PIN but it's not on the sensor(BD_Sensor Attached)

SKR V1.3 Pins file
#define I2C_BD_SDA_PIN  P1_24 // Change to actual number which the SDA wire is connected to on your mainboard Z_MAX
(Will change to Z_MIN if code doesn't like Z_MAX as the pin and then use sensor for auto homing.)
#define I2C_BD_SCL_PIN  P2_00 // Change to actual number which the SCL wire is connected to on your mainboard
#define I2C_BD_DELAY  20     // Defeult value is 20, should be in range [20, 50]

I changed all the wires according to the sensor and put the SCL where the CLK plugs in. Didn't smoke when turned on, the sensor only blinked a green light. Hope this means ok to continue with the setup. Hope it doesn't smoke it.

[image]
[image]

markniu, 2024-12-30:
right, the SCL_PIN means the CLK pin.
there should some version string after M102 S-1 , else it's working.
and I just checked this skr v1.3 , there are signal filter capacitors on all the endstop pins, so we can not use those pins
[image]

please choose other pins like the EXP1 EXP2 or TFT

brosskgm, 2024-12-30:
Hi Mark, Thanks for the quick reply.

Do you have any instructions for the EXP1 or EXP2. Already using the TFT pins for the screen.

Thanks
Bob

markniu, 2024-12-31:
are there some message returned after M102 S-1  like `pandapi3d.com v1.2c`

brosskgm, 2025-01-01:
M102 S-1
Returns OK

brosskgm, 2025-01-04:
Hi Mark,

I decided to get the SKR V1.4 board since it has the Survo and Probe pins seperate. Will get in touch with you then.

Thanks
Bob

brosskgm, 2025-01-12:
Hi Mark,

Got the SKR V1.4 all installed. Now when I send the M102 S-1 Command it gives unknown command.

markniu, 2025-01-13:
I just check that the gcode M102 S-1 and some features are not working in the  marlin mainline, could you use this https://github.com/markniu/MarlinPULL  that I have just updated to the latest marlin.

brosskgm, 2025-01-13:
Went through and changed the Config and Pins files and now when I try to compile it gives me the attached error that it's not a Valid Marlin 2.x project.

[image]

markniu, 2025-01-14:
https://github.com/MarlinFirmware/Marlin/issues/23431#issuecomment-1003749157

> Went through and changed the Config and Pins files and now when I try to compile it gives me the attached error.
> 
> ![VisualError](https://private-user-images.githubusercontent.com/3050016/402699577-eaf2d2a2-00e8-4685-a859-571b3e164964.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY4MTkyMDYsIm5iZiI6MTczNjgxODkwNiwicGF0aCI6Ii8zMDUwMDE2LzQwMjY5OTU3Ny1lYWYyZDJhMi0wMGU4LTQ2ODUtYTg1OS01NzFiM2UxNjQ5NjQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDExNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMTRUMDE0MTQ2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTYwZmM4NmZiNGJjOWEyYzhkZTBmODMxNmMwM2I1MWExN2MxMDI0ZWMwYmY1NmM1YWUwYzk3M2Y1YWIyMTA4OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.vWOJqho4aDkiWmy0DsdKXln95jF8wrv7gfQJhZ-AXeg)

brosskgm, 2025-01-14:
I'll check. This was on the SKR V1.4 - It compikles fine with all the other Marlin directories. Even when I select to open directory and select it, it said the same erro. I downloaded 2.1.2.5 bug fix and it didn't give the error. I'll see if I can make the config changes in the bug fix and see if it starts to work.

brosskgm, 2025-01-14:
I had to take all the dashes(-) and dots out of the folder name it it accepted it.

The compile now didn't like it. 89 errors.

[image]
[image]
[image]
[image]

brosskgm, 2025-01-14:
After clearing the problems/errors it now only fails the build with the same error as above.

Marlin/src/module/probe.cpp: In static member function 'static float Probe::probe_at_point(const_float_t, const_float_t, ProbePtRaise, uint8_t, bool, bool, const_float_t, const_float_t, bool)':
Marlin/src/module/probe.cpp:995:43: error: 'class BDS_Leveling' has no member named 'read'; did you mean 'bd_read'?
  995 |     measured_z = current_position.z - bdl.read(); // Difference between Z-home-relative Z and sensor reading
      |                                           ^~~~
      |                                           bd_read
Compiling .pio/build/LPC1768/src/src/module/stepper/indirection.cpp.o
*** [.pio/build/LPC1768/src/src/module/probe.cpp.o] Error 1

markniu, 2025-01-15:
yes, please change the` bdl.read();` to `bdl.bd_read();` in the probe.cpp,  that I have updated to the [pull](https://github.com/MarlinFirmware/Marlin/pull/27243/files#diff-844640fa2c3026003779111f0d4594b9370678140e782a408409d57a9c1dc809) too

> After clearing the problems/errors it now only fails the build with the same error as above. Here is the line in probe.cpp it's talking about and photo from the file.
>

brosskgm, 2025-01-15:
It compiled. Will run it on the printer tomorrow. Will let you know how it does.

brosskgm, 2025-01-15:
It all worked. Looks like I'll need to work on the sensor heigh.

[image]
[image]
[image]
[image]

It doesn't touch the bed, but looks like I will need to find a way to raise it.

[image]

brosskgm, 2025-01-15:
Or maby lower it. The Mic say little over 4mm height.

markniu, 2025-01-15:
right, the first calibration data is too high, We can simply calculate the sensor to the bed is about 629*0.005 = 3.14 mm ,

brosskgm, 2025-01-15:
The mount is the longest one on the three in the stl files.

brosskgm, 2025-01-15:
I also printed the BL-Touch mount. Might have to see if I can find a couple with different heigths.

brosskgm, 2025-01-15:
Yep. Found one 3mm shorter. Will print it up tomorrow and re-do it all.

brosskgm, 2025-01-15:
Printed a new BLTouch mount. Extended the height 1.5mm and after calibration M102 S-5 if doesn't give the Bad Sensor height.

Maybe since I have the probe in the front of nozzle, might not really need to worry if I have it go to the regular home location 0 as before probe was installed.

[image]

Getting closer.

markniu, 2025-01-16:
From the raw calibrate data, it seems that you need to check the steps of the Z axis. 
if the first point is 149 then second data should be at least about 149  + 20 = 169.

brosskgm, 2025-01-16:
Not sure where to find that one. The Z_STEPS in configuration.h are only for ABL or MBL and not enabled.

brosskgm, 2025-01-16:
Could only find AXIS steps.

[image]

brosskgm, 2025-01-16:
Here is what ahppens when G28 is sent. It keeps re-homing again over and over. The only way to stop it is turn off the printer.

brosskgm, 2025-01-16:
Here is what ahppens when G28 is sent. It keeps re-homing again over and over. The only way to stop it is turn off the printer. Never get the chance to send G1 Z0

[image]

closed 2025-01-16 by brosskgm

brosskgm, 2025-01-17:
Funny they say I closed this yesterday and most was deleted. I didn't log in yesterday at all. I'll keep trying to figure it out.