markniu/Bed_Distance_sensor #205: unable to compile into klipper

mataksik, 2024-12-30:
[image]
[image]
[image]
[image]
[klippy (1).log](https://github.com/user-attachments/files/18278388/klippy.1.log)

Eryone, 2024-12-31:
Which MCU board do you use?    you can also check if the boot option of the mcu board is right.

markniu, 2024-12-31:
what's your mcu board?  you can also check again the boot option of the mcu board. check the address of the boot option is right to the 0x8002000

mataksik, 2024-12-31:
was happy using BD Sensor with bigtreetech CB1 on MANTA M4P board  for a year.
Realy best probe ever 
until UPDATE... few new sd cards, new cb1 images, can't get it back running

mcu 'mcu': Starting serial connect
webhooks client 281473863725776: New connection
webhooks client 281473863725776: Client info {'program': 'Moonraker', 'version': 'v0.9.3-4-ga4604e3'}
Loaded MCU 'mcu' 118 commands (v0.12.0-404-g80d185c94 / gcc: (15:12.2.rel1-1) 12.2.1 20221205 binutils: (2.40-2+18+b1) 2.40)
MCU 'mcu' config: ADC_MAX=4095 BUS_PINS_i2c1_PA9_PA10=PA9,PA10 BUS_PINS_i2c1_PB6_PB7=PB6,PB7 BUS_PINS_i2c1_PB8_PB9=PB8,PB9 BUS_PINS_i2c2_PB10_PB11=PB10,PB11 BUS_PINS_i2c2_PB13_PB14=PB13,PB14 BUS_PINS_i2c3_PB3_PB4=PB3,PB4 BUS_PINS_i2c3_PC0_PC1=PC0,PC1 BUS_PINS_spi1=PA6,PA7,PA5 BUS_PINS_spi1a=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13 BUS_PINS_spi2_PB2_PB11_PB10=PB2,PB11,PB10 BUS_PINS_spi2a=PC2,PC3,PB10 BUS_PINS_spi3=PB4,PB5,PB3 CLOCK_FREQ=64000000 MCU=stm32g0b1xx PWM_MAX=255 RESERVE_PINS_USB=PA11,PA12 RESERVE_PINS_crystal=PF0,PF1 STATS_SUMSQ_BASE=256 STEPPER_BOTH_EDGE=1
Protocol error
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/klippy.py", line 135, in _connect
    cb()
  File "/home/biqu/klipper/klippy/mcu.py", line 748, in _connect
    self._send_config(None)
  File "/home/biqu/klipper/klippy/mcu.py", line 679, in _send_config
    cb()
  File "/home/biqu/klipper/klippy/extras/BDsensor.py", line 716, in build_config
    self.I2C_BD_send_cmd = self.mcu.lookup_query_command(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/biqu/klipper/klippy/mcu.py", line 883, in lookup_query_command
    return CommandQueryWrapper(self._serial, msgformat, respformat, oid,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/biqu/klipper/klippy/mcu.py", line 56, in __init__
    self._cmd = serial.get_msgparser().lookup_command(msgformat)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/biqu/klipper/klippy/msgproto.py", line 314, in lookup_command
    self._error("Unknown command: %s", msgname)
  File "/home/biqu/klipper/klippy/msgproto.py", line 242, in _error
    raise error(self.warn_prefix + (msg % params))
msgproto.error: mcu 'mcu': Unknown command: I2CBD
MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-404-g80d185c94
MCU(s) which should be updated:
<none>
Up-to-date MCU(s):
mcu: Current version v0.12.0-404-g80d185c94

mataksik, 2025-01-01:
in 19th attempt managed to get working. works only when not using **./make_with_bdsensor.sh**

cd ~/klipper/
make menuconfig
git status
git reset --hard origin/master
git pull
cd ~/Bed_Distance_sensor/klipper/
./install_BDsensor.sh
cd ~/klipper
make clean
make
make flash FLASH_DEVICE=/dev/serial/by-id/usb-Klipper_stm32g0b1xx_590034000C50415833323420-if00
sudo service klipper restart

Happy New Year, Happy Printing