markniu/Bed_Distance_sensor #207: Bed Distance Sensor data error2:10.24

abusaud64, 2025-01-09:
Hello Mark

This my printer
Delta ( Mini Kossel )
Main Board: BTT E3 SKR MINI V3.0 Main Board
Host: Fly pi v2
CanBus: EBB42 V1.2
BD Sensor V1.3

I have successfully compiled Klipper on all the components mentioned, and the printer is working great. However, I am now encountering an error message: "**Bed Distance Sensor data error2:10.24.**"

I’ve tried all possible solutions, but none have worked. Then, I came across this link: [](https://github.com/markniu/Bed_Distance_sensor/issues/117#issuecomment-2062839664.)

I attempted the suggested solution on the Fly Pi V2, but I received the error: "**Unknown command: I2CBD**."

Honestly, I was very optimistic about this sensor and was planning to use it in all of my printers, but now I’m quite disappointed. What do you recommend I do?

markniu, 2025-01-09:
Which port and gpio was the BDsensor connected to?

markniu, 2025-01-09:
For the EBB42, you can try to use the I2C port, and set the delay value 20.
[BDsensor]
sda_pin: EBBCan: PB3
scl_pin: EBBCan: PB4
delay:20

Unknown command: I2CBD means the code in BD_sensor.c was not compiled into the firmware or the firmware was not flashed success.

markniu, 2025-01-10:
I just tested on my EBB36 today, it works well. maybe you can check again the connection, and the connection should like this.
[image]
[image]
[image]

BTW, please use the default cable to test at first, because there are some data transfer on the cable like the usb cable required more stable and high quality cable

abusaud64, 2025-01-10:
Victory! The issue has been resolved, and everything is running smoothly now. 

Thank you so much, Mr. Mark. Your perseverance, dedication, and determination are truly admirable. I deeply appreciate your unwavering effort in helping me overcome this challenge, and I’m grateful for how you restored my hope. Thank you again for everything!

abusaud64, 2025-01-10:
Sorry Mark

 probe is always showing as "TRIGGERED."

[klippy (7).log](https://github.com/user-attachments/files/18376745/klippy.7.log)

15:21:02 $ QUERY_PROBE
15:21:03 // probe: TRIGGERED

15:22:12 $ M119
15:22:13 a:TRIGGERED b:TRIGGERED c:TRIGGERED

15:22:18 $ M102 S-1
15:22:18 // No data or corrupt data from BDsensor(                    ), Please check connection

15:22:24 $ M102   S-2
15:22:24 0.0mm

markniu, 2025-01-10:
The EBB can not communicate to the sensor because there is no data after M102 S-1 

which cable do you use, could you take a picture of that.

> Sorry Mark
> 
> Ok No more Bed Distance Sensor data error2:10.24 But probe is always showing as "TRIGGERED."
> 
> [klippy (7).log](https://github.com/user-attachments/files/18376745/klippy.7.log)
> 
> 15:21:02 $ QUERY_PROBE 15:21:03 // probe: TRIGGERED
> 
> 15:22:12 $ M119 15:22:13 a:TRIGGERED b:TRIGGERED c:TRIGGERED
> 
> 15:22:18 $ M102 S-1 15:22:18 // No data or corrupt data from BDsensor( ), Please check connection
> 
> 15:22:24 $ M102 S-2 15:22:24 0.0mm

abusaud64, 2025-01-10:
I used the original cable that came with the sensor.

[image]

markniu, 2025-01-10:
this is the old type of cable that may have issue. and even cannot be detected by the multimeter. I would like to send you the new one, please send your shipping address to my email:niujl123@sina.com

abusaud64, 2025-01-10:
> [abusaud64](/abusaud64)

I sent you from my mail  **Protonmail.com**

abusaud64, 2025-01-27:
Thanks, Mark. I received the new one, and everything works great

GoTVm, 2025-03-09:
Hello, I started having this issue today as well, sensor doesn't work anymore and it reads triggered at all times. What's with the cable?

markniu, 2025-03-10:
did you enable the collision_sensing? the endstop read command M119 will not work if the collision_sensing is enabled.

> Hello, I started having this issue today as well, sensor doesn't work anymore and it reads triggered at all times. What's with the cable?

GoTVm, 2025-03-10:
Yes, it has always been enabled, even before when it was working. It just started acting up all of a sudden.