markniu/Bed_Distance_sensor #208: Nozzle crashes to the bed

Nmorelm, 2025-01-16:
Suddenly the nozzle crashed to the bed.
First time after replaced my printer bed ,everything was ok. I printed some stuff. Later ive seen that first layer isnt perfect. So i messed with that and at some point nozzle just crashed to the bed. 
I checked with metal ruler if it triggers and even tried Z homing and put metal ruler under the sensor - its stopped.
Later , then replaced broken parts of the bed arms, i wanted to calibrate the sensor. I homed X and Y, manually got Z to zero and sent M102 S-6 command. To that command it started to home X and Y again (why?) and then just fully crashed to bed ..again.
Any ideas what is wrong ?

markniu, 2025-01-16:
you can check the connection by sending M102 S-1 to see if there is a string about version returned

Nmorelm, 2025-01-16:
What could i do / check next ?
Just checked again, if i manually move Z to zero , sensor shows triggered.
M102   S-2 reports 0.22mm
But when move Z back to 1mm , M102   S-2 reports 0.22mm again. But i not homed Z yet, as i dont want to break parts again.

With Z manually set to 0 , M102 S-5 reports :
`
21:51:14
846 at 3.9mm
21:51:14
843 at 3.8mm
21:51:14
839 at 3.7mm
21:51:14
835 at 3.6mm
21:51:13
831 at 3.5mm
21:51:13
826 at 3.4mm
21:51:13
822 at 3.3mm
21:51:13
817 at 3.2mm
21:51:13
813 at 3.1mm
21:51:13
807 at 3.0mm
21:51:13
803 at 2.9mm
21:51:13
798 at 2.8mm
21:51:13
790 at 2.7mm
21:51:13
783 at 2.6mm
21:51:13
775 at 2.5mm
21:51:13
767 at 2.4mm
21:51:13
759 at 2.3mm
21:51:13
752 at 2.2mm
21:51:13
742 at 2.1mm
21:51:13
733 at 2.0mm
21:51:13
723 at 1.9mm
21:51:13
711 at 1.8mm
21:51:13
699 at 1.7mm
21:51:13
689 at 1.6mm
21:51:13
677 at 1.5mm
21:51:13
662 at 1.4mm
21:51:13
649 at 1.3mm
21:51:13
634 at 1.2mm
21:51:13
619 at 1.1mm
21:51:13
604 at 1.0mm
21:51:13
585 at 0.9mm
21:51:13
568 at 0.8mm
21:51:12
550 at 0.7mm
21:51:12
531 at 0.6mm
21:51:12
509 at 0.5mm
21:51:12
490 at 0.4mm
21:51:12
467 at 0.3mm
21:51:12
445 at 0.2mm
21:51:12
423 at 0.1mm
21:51:12
403 at 0.0mm`

What means that sensor is too high form bed, but it less then 1mm.

[image]

markniu, 2025-01-17:
you can send M102 S-7 to get the raw value at different distance to see if all the hardware of sensor work.

```
SET_KINEMATIC_POSITION Z=5
```
then move the nozzle down or up to close to the bed, send M102 S-7

Nmorelm, 2025-01-17:
With Z=0  Raw data:21,  Then we can calculate the distance by comparing to the calibration data:0.22mm
With Z=1  Raw data:237,  Then we can calculate the distance by comparing to the calibration data:0.22mm
With Z=2  Raw data:480,  Then we can calculate the distance by comparing to the calibration data:0.58mm
With Z=3  Raw data:659,  Then we can calculate the distance by comparing to the calibration data:1.5999999999999999mm
With Z=4  Raw data:766,  Then we can calculate the distance by comparing to the calibration data:2.6100000000000003mm
With Z=5  Raw data:829,  Then we can calculate the distance by comparing to the calibration data:3.68mm
With Z=6  Raw data:868,  Then we can calculate the distance by comparing to the calibration data:4.12mm

Back to Z=1 Raw data:242,  Then we can calculate the distance by comparing to the calibration data:0.22mm

Hope this helps!

markniu, 2025-01-17:
Seems the sensor works. I think the only thing need to do is to calibrate the sensor again.

if the `collision_calibrate:1` then it will run G28 at first after sending M102 S-6 command and no need to manual move the z axis
if the `collision_homing:1` , it will do collision homing at each G28,like this
https://www.youtube.com/watch?v=Cdexsotm0Sc

> Suddenly the nozzle crashed to the bed. First time after replaced my printer bed ,everything was ok. I printed some stuff. Later ive seen that first layer isnt perfect. So i messed with that and at some point nozzle just crashed to the bed. I checked with metal ruler if it triggers and even tried Z homing and put metal ruler under the sensor - its stopped. Later , then replaced broken parts of the bed arms, i wanted to calibrate the sensor. I homed X and Y, manually got Z to zero and sent M102 S-6 command. To that command it started to home X and Y again (why?) and then just fully crashed to bed ..again. Sensor installed around 1mm from the bed , it not changed. It was running fine and gave nice calibration results before.
> 
> Any ideas what is wrong ?

Nmorelm, 2025-01-17:
> Seems the sensor works. I think the only thing need to do is to calibrate the sensor again.
>
> if the `collision_calibrate:1` then it will run G28 at first after sending M102 S-6 command and no need [...]

Last time it just crashed to the bed after homing X and Y

markniu, 2025-01-17:
yes, you can try that, set the `collision_calibrate:0` and `collision_homing:0`

Nmorelm, 2025-01-17:
Ok so i did it with collision_calibrate:0 and collision_homing:0 , calibrated sensor and then did with collision_calibrate:1 and collision_homing:1 , calibrated again.
Successfuly.
Thank you for your time.

closed 2025-02-03 by Nmorelm