markniu/Bed_Distance_sensor #209: Bed Distance Sensor data error2:10.24

PongsakBS, 2025-01-16:
Dear Mark

My systems details as below
-Main Board: BTT SKR 3 EZ. Main Board
-Host: Raspberry pi 4 Ram 8
-CanBus: EBB36 V1.2
-BD Sensor V1.3

I have run the BD Sensor smoothly since few day ago , currently when I press homing then I got the error **Bed Distance Sensor data error2:10.24**  
Then I check communication via **BDSENSOR_VERSION** they are still working.
You can see the screen shot below for error msg and my configuration. Could you please advise the solution to fix it?

[image]

markniu, 2025-01-17:
take a picture of your cable that connect to the bdsensor.  maybe there is a connection problem.

markniu, 2025-01-17:
take a picture of your cable that connect to the bdsensor. maybe there is a connection problem in the connector or in the cable.

PongsakBS, 2025-01-17:
Dear Mark 

the cable is find due to I can check the BD version. 

Now it fixed , I did re calibrate again 

**5. Calibration**
Clean the nozzle

Send command SET_KINEMATIC_POSITION Z=200  then move the Z axis down in the menu until the nozzle just touches the bed plate, the BDsensor will use this position as the zero position.

Sending gcode command **M102 S-6.**

After that you can check whether the BDsensor has been calibrated successful by M102 S-5 that will return the raw calibration data which stored in the BDsensor. If the first raw calibration data returned by the M102 S-5 is greater than 400, that means the sensor is mounted too high and needs to be remounted closer to the bed, the recommend value of first data is around 100. also make sure that the value of second data is greater than the first data by more than 10, else adjust the height of z and do calibarte again.