markniu/Bed_Distance_sensor #210: BDSensor V1.1 on EBB36 V1.2 CAN Board No data or corrupt data from BDsensor()

Qtronics-Electronics, 2025-01-19:
BD Sensor V1.1 (Version is off of BDSensor Board)
EBB36 V1.2 

Verified that firmware was loaded, and EBB36 is showing up on CAN Bus. All other functions of the board are working. 

M102 S-1
No data or corrupt data from BDsensor(), Please check connection

Klipper Settings are 
[BDsensor] 
sda_pin: EBBCan:PB8
scl_pin: EBBCan:PB9
delay: 10 # you can set it 10 if the BDsensor version is >=1.2 # (Tried several different values here)
z_offset:0 #  within -0.6 to 0.6mm 
x_offset: -34
y_offset: 0

Flashed EBB36 using instruction from https://github.com/markniu/Bed_Distance_sensor/wiki/Installing-for-Klipper

Confirmed that out/src/BD_sensor.o was included in make

I also tried on a second board with the exact same issue.

Light on BDsensor is working as expected. 

[image]

markniu, 2025-01-20:
the `delay `value should be set to 20 in the BDsensor section

Qtronics-Electronics, 2025-01-20:
I tried several values in the delay and nothing worked. I also just double-checked with 20 and still get the same error.

markniu, 2025-01-20:
this old type of cable may have a issue.  I think you can rearrange the position of cables by tweezers:
for example,  switch the yellow <-->green,  orange<-->grey

Qtronics-Electronics, 2025-01-20:
Here is how it is wired to the EBB36.

Pin Name on BDSensor  --- Wire Color --- Pin on EBB36

SDA --- (Green) --- PB8
GND --- (Yellow) --- GND
CLK --- (Orange) --- PB9
5V --- (Red) --- 5V
GND --- (Brown)  --- GND

Do you mean change pinout to 

PB8 --- Yellow
GND --- Green
CLK --- Brown
GND --- Orange
5V --- Red

I am checking to make sure I dont blow something up.

markniu, 2025-01-20:
yes, just rearrange the position of wires :

SDA --- (Yellow) --- PB8
GND --- (Green) --- GND
CLK --- (Brown) --- PB9
5V --- (Red) --- 5V
GND --- (Orange) --- GND

Qtronics-Electronics, 2025-01-20:
I tried it, and I am getting the same issue.

Qtronics-Electronics, 2025-01-20:
I also notice the the light on the BD Sensor quit working. I switched Brown and orange back and the light started working but I still get the failure.

markniu, 2025-01-20:
that means the sensor is working except the communication, actually not sure where is the connection problem.  but I can send you a new sensor and cable, please send me (markyueyue@gmail.com)  your shipping address.
> I also notice the the light on the BD Sensor quit working. I switched Brown and orange back and the light started working but I still get the failure.

QTronics, 2026-01-19:
> Was this fixed with a new sensor? I have the same problem as described here but with a EBB36 Gen2.
 
Yes a new sensor did correct the issue.