markniu/Bed_Distance_sensor #211: Auto bed calibration conflict with modified G28 macro witch results Macro G28 called recursively

syrekszymon, 2025-01-29:
The recently added feature "Auto bed calibration" is causing a conflict with [modified G28 macro](https://github.com/markniu/Bed_Distance_sensor/wiki/Homing-with-contactless-and-contact-probing-together) which results in the following error: 

`17:54
Macro G28 called recursively
17:54
Macro G28 called recursively
17:54
Macro G28 called recursively
17:54
Macro G28 called recursively
17:54
Homing z
17:54
Detect the plate changed or others, the BD sensor needs recalibration -0.060
17:54
bd_value at 0.440 mm
17:54
Raw data:283 at 0 mm, BDsensor to bed: 1.1320 mm, Bed: 38.9C, z_offset: 0.000
17:53
Homing_speed at 2.000
17:53
BDsensorVer:V1.2c,switch_mode=1,collision_homing=0,collision_cal=1
17:53
homing all
17:53
G28`

markniu, 2025-01-30:
Thanks for the feedback.
just fixed [this error](https://github.com/markniu/Bed_Distance_sensor/commit/2d5207e7c2e8046da372b8fd0f336da09fee0b98)
you can update the BDsensor.py and restart klipper again.
also need to add  `homing_cmd:G990028`  into the section `[BDsensor] `

syrekszymon, 2025-01-30:
Fixed, thx.

closed 2025-01-30 by syrekszymon

mikemerryguy, 2025-03-02:
maybe add this to the main page to let people know!, 

set up an error catcher that doesn't allow it to auto calibrate without the homing_cmd 

I just destroyed a BRAND NEW PEI SHEET bc this wasn't address properly.

when I found this I thought it was awesome, I had no idea it was a BETA sensor, getting a little ticked off as of late with it....

markniu, 2025-03-04:
Thanks for remind me, and I just added into the FAQ and the default section [BDsensor]

> maybe add this to the main page to let people know!,
> set up an error catcher that doesn't allow it to auto calibrate without the homing_cmd
> I just destroyed a BRAND NEW PEI SHEET bc this wasn't [...]