markniu/Bed_Distance_sensor #213: No trigger on z after full movement

MDegelmann, 2025-02-08:
I recently had the problem that I can no longer home. I always get the same error

[image]

markniu, 2025-02-08:
could you attach the klippy.log file?  there will be more information in that file

markniu, 2025-02-08:
you can also try to recalibrate the bdsensor

MDegelmann, 2025-02-08:
`Starting Klippy...
Args: ['/home/biqu/klipper/klippy/klippy.py', '/home/biqu/printer_data/config/printer.cfg', '-I', '/home/biqu/printer_data/comms/klippy.serial', '-l', '/home/biqu/printer_data/logs/klippy.log', '-a', '/home/biqu/printer_data/comms/klippy.sock']
Git version: 'v0.12.0-432-gfec3e685c-dirty'
Untracked files: klippy/extras/BDsensor.py
Branch: master
Remote: origin
Tracked URL: https://github.com/Klipper3d/klipper
CPU: 4 core ?
Python: '3.9.2 (default, Dec  1 2024, 12:12:57) \n[GCC 10.2.1 20210110]'
Start printer at Sat Feb  8 09:54:29 2025 (1739008469.5 2768.1)
===== Config file =====
[virtual_sdcard]
path = ~/printer_data/gcodes
on_error_gcode = CANCEL_PRINT

[pause_resume]

[display_status]

[respond]

[gcode_macro CANCEL_PRINT]
description = Cancel the actual running print
rename_existing = CANCEL_PRINT_BASE
gcode = 
	
	{% set client = printer['gcode_macro _CLIENT_VARIABLE']|default({}) %}
	{% set allow_park = client.park_at_cancel|default(false)|lower == 'true' %}
	{% set retract = client.cancel_retract|default(5.0)|abs %}
	
	{% set park_x = "" if (client.park_at_cancel_x|default(none) is none)
	else "X=" ~ client.park_at_cancel_x %}
	{% set park_y = "" if (client.park_at_cancel_y|default(none) is none)
	else "Y=" ~ client.park_at_cancel_y %}
	{% set custom_park = park_x|length > 0 or park_y|length > 0 %}
	
	
	{% if printer['gcode_macro RESUME'].restore_idle_timeout > 0 %}
	SET_IDLE_TIMEOUT TIMEOUT={printer['gcode_macro RESUME'].restore_idle_timeout}
	{% endif %}
	{% if (custom_park or not printer.pause_resume.is_paused) and allow_park %} _TOOLHEAD_PARK_PAUSE_CANCEL {park_x} {park_y} {% endif %}
	_CLIENT_RETRACT LENGTH={retract}
	TURN_OFF_HEATERS
	M106 S0
	{client.user_cancel_macro|default("")}
	SET_GCODE_VARIABLE MACRO=RESUME VARIABLE=idle_state VALUE=False
	
	SET_PAUSE_NEXT_LAYER ENABLE=0
	SET_PAUSE_AT_LAYER ENABLE=0 LAYER=0
	CANCEL_PRINT_BASE

[gcode_macro PAUSE]
description = Pause the actual running print
rename_existing = PAUSE_BASE
gcode = 
	
	{% set client = printer['gcode_macro _CLIENT_VARIABLE']|default({}) %}
	{% set idle_timeout = client.idle_timeout|default(0) %}
	{% set temp = printer[printer.toolhead.extruder].target if printer.toolhead.extruder != '' else 0 %}
	{% set restore = False if printer.toolhead.extruder == ''
	else True  if params.RESTORE|default(1)|int == 1 else False %}
	
	SET_GCODE_VARIABLE MACRO=RESUME VARIABLE=last_extruder_temp VALUE="{{'restore': restore, 'temp': temp}}"
	
	{% if idle_timeout > 0 %}
	SET_GCODE_VARIABLE MACRO=RESUME VARIABLE=restore_idle_timeout VALUE={printer.configfile.settings.idle_timeout.timeout}
	SET_IDLE_TIMEOUT TIMEOUT={idle_timeout}
	{% endif %}
	PAUSE_BASE
	{client.user_pause_macro|default("")}
[... 3047 lines cut ...]
Receive: 80 2807.632528 2807.631712 14: seq: 13, trsync_state oid=12 can_trigger=1 trigger_reason=0 clock=2099428096
Receive: 81 2807.647461 2807.636727 14: seq: 14, trsync_state oid=12 can_trigger=1 trigger_reason=0 clock=2102428056
Receive: 82 2807.663795 2807.636727 14: seq: 14, trsync_state oid=12 can_trigger=1 trigger_reason=0 clock=2105428076
Receive: 83 2807.663844 2807.636727 14: seq: 14, trsync_state oid=12 can_trigger=1 trigger_reason=0 clock=2108428071
Receive: 84 2807.663957 2807.636727 14: seq: 14, trsync_state oid=12 can_trigger=0 trigger_reason=4 clock=2110910071
Receive: 85 2807.663967 2807.636727 14: seq: 14, trsync_state oid=12 can_trigger=0 trigger_reason=4 clock=2111427939
Receive: 86 2807.663979 2807.663351 10: seq: 15, trsync_state oid=12 can_trigger=0 trigger_reason=4 clock=0
Receive: 87 2807.666888 2807.666470 10: seq: 16, trsync_state oid=12 can_trigger=0 trigger_reason=4 clock=0
Receive: 88 2807.668597 2807.668131 10: seq: 17, stepper_position oid=11 pos=-18653
Receive: 89 2807.674560 2807.670260 18: seq: 18, tmcuart_response oid=2 read=b'\n\xfaO-\x80\x00\x02h?\x93'
Receive: 90 2807.678431 2807.678012 10: seq: 19, stepper_position oid=13 pos=-18653
Receive: 91 2807.684567 2807.680173 18: seq: 1a, tmcuart_response oid=3 read=b'\n\xfaO-\x80\x00\n\xe8\xa1\xae'
Receive: 92 2807.686518 2807.686110 10: seq: 1b, stepper_position oid=14 pos=18653
Receive: 93 2807.692489 2807.688234 18: seq: 1c, tmcuart_response oid=4 read=b'\n\xfaO-\x80\x00\x02\xe8\xa9\xf2'
Receive: 94 2807.766310 2807.695107 15: seq: 1d, analog_in_state oid=15 next_clock=2270065408 value=31624
Receive: 95 2808.066322 2807.933573 15: seq: 1e, analog_in_state oid=15 next_clock=2390065408 value=31625
Receive: 96 2808.366307 2807.933573 15: seq: 1e, analog_in_state oid=15 next_clock=2510065408 value=31624
Receive: 97 2808.666305 2807.933573 15: seq: 1e, analog_in_state oid=15 next_clock=2630065408 value=31624
Receive: 98 2808.716413 2808.716025 11: seq: 1f, clock clock=2532971606
Receive: 99 2808.773789 2808.773171 12: seq: 10, shutdown clock=2555873675 static_string_id=Command request
Requested toolhead position at shutdown time 27.864521: (47.0, 55.0, 1.2)
Stats 2809.7: gcodein=0   mcu: mcu_awake=0.017 mcu_task_avg=0.000002 mcu_task_stddev=0.000001 bytes_write=21772 bytes_read=27367 bytes_retransmit=0 bytes_invalid=0 send_seq=1409 receive_seq=1409 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=399999033 canstat_EBBCan: bus_state=active rx_error=0 tx_error=0 tx_retries=0 EBBCan: mcu_awake=0.055 mcu_task_avg=0.000018 mcu_task_stddev=0.000041 bytes_write=9514 bytes_read=19409 bytes_retransmit=0 bytes_invalid=0 send_seq=852 receive_seq=852 retransmit_seq=0 srtt=0.001 rttvar=0.001 rto=0.02 [... cut ...] ate=active rx_error=0 tx_error=0 tx_retries=0 EBBCan: mcu_awake=0.031 mcu_task_avg=0.000018 mcu_task_stddev=0.000037 bytes_write=9538 bytes_read=19592 bytes_retransmit=0 bytes_invalid=0 send_seq=856 receive_seq=856 retransmit_seq=0 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=64001219 adj=64001519  heater_bed: target=0 temp=19.3 pwm=0.000 sysload=3.06 cputime=12.535 memavail=627028 print_time=27.371 buffer_time=0.000 print_stall=0 extruder: target=0 temp=19.5 pwm=0.000
`

MDegelmann, 2025-02-08:
> Sie können auch versuchen, den BD-Sensor neu zu kalibrieren

Habe ich mehrfach ohne erfolg

MDegelmann, 2025-02-08:
[klippy.log](https://github.com/user-attachments/files/18717889/klippy.log)

markniu, 2025-02-08:
What's the distance (in mm) to lift the Z axis prior to homing like the z_hop value?
I can't find that value in the log file.
or you can a section [safe_z_home] [https://www.klipper3d.org/Config_Reference.html#safe_z_home](https://www.klipper3d.org/Config_Reference.html#safe_z_home)   and set the z_hop value at least 5

closed 2025-02-11 by MDegelmann