markniu/Bed_Distance_sensor #215: Connection issue and bad calibration  values

rmbq, 2025-02-16:
Hi,
i'm throubleshoting my new BDsensor.
i noticed the red LED was blinking slowly and no data was ever read. Push the connector hard with my finger made it to blink fast:
https://github.com/user-attachments/assets/00f51a41-f429-461e-ac9e-f06c8bc775b4

I made a calibration this way, but i don't think get values are good enough:

> 10:44:23.684 M102 S-5
> 10:44:23.689 ok
> 10:44:23.826 Calibrate data:0,47
> 10:44:23.879 Calibrate data:1,64
> 10:44:23.926 Calibrate data:2,65
> 10:44:23.976 Calibrate data:3,10
> 10:44:24.026 Calibrate data:4,47
> 10:44:24.076 Calibrate data:5,19
> 10:44:24.126 Calibrate data:6,12
> 10:44:24.176 Calibrate data:7,199
> 10:44:24.226 Calibrate data:8,10
> 10:44:24.280 Calibrate data:9,46
> 10:44:24.326 Calibrate data:10,23
> 10:44:24.376 Calibrate data:11,45
> 10:44:24.426 Calibrate data:12,14
> 10:44:24.476 Calibrate data:13,24
> 10:44:24.526 Calibrate data:14,46
> 10:44:24.576 Calibrate data:15,13
> 10:44:24.626 Calibrate data:16,57
> 10:44:24.676 Calibrate data:17,13
> 10:44:24.726 Calibrate data:18,40
> 10:44:24.776 Calibrate data:19,43
> 10:44:24.826 Calibrate data:20,24
> 10:44:24.876 Calibrate data:21,40
> 10:44:24.926 Calibrate data:22,15
> 10:44:24.976 Calibrate data:23,53
> 10:44:25.026 Calibrate data:24,13
> 10:44:25.076 Calibrate data:25,41
> 10:44:25.126 Calibrate data:26,10
> 10:44:25.176 Calibrate data:27,43
> 10:44:25.226 Calibrate data:28,12
> 10:44:25.276 Calibrate data:29,43
> 10:44:25.326 Calibrate data:30,20
> 10:44:25.376 Calibrate data:31,35
> 10:44:25.426 Calibrate data:32,13
> 10:44:25.476 Calibrate data:33,13
> 10:44:25.526 Calibrate data:34,42
> 10:44:25.576 Calibrate data:35,22
> 10:44:25.626 Calibrate data:36,40
> 10:44:25.676 Calibrate data:37,12
> 10:44:25.726 Calibrate data:38,53
> 10:44:25.776 Calibrate data:39,13

And using `M102 S-2` at the same height give me different results:

> 10:46:10.087 Bed Distance:0.01mm
> 10:46:10.087 ok
> 10:46:11.023 M102 S-2
> 10:46:11.033 Bed Distance:2.30mm
> 10:46:11.033 ok
> 10:46:12.033 wait
> 10:46:12.135 M102 S-2
> 10:46:12.145 Bed Distance:1.50mm
> 10:46:12.145 ok
> 10:46:12.926 M102 S-2
> 10:46:12.939 Bed Distance:2.20mm
> 10:46:12.939 ok
> 10:46:13.757 M102 S-2
> 10:46:13.767 Bed Distance:1.32mm
> 10:46:13.767 ok
> 10:46:14.712 M102 S-2
> 10:46:14.720 Bed Distance:0.09mm

Without the finger pushing all the values are 1015.
version is `1.2c`
can you help me?

markniu, 2025-02-16:
Hello, It seems that your nozzle is not moving up and live away the bed while calibrating.
also you can test this sensor without calibration by gcode ` M102 S-7` to get the raw data in different distance.

rmbq, 2025-02-16:
But the nozzle is moving away from the bed, I can clearly see every step of the z axis. I will try to take a look at the raw values with S-7

And what about the connector issue?

markniu, 2025-02-16:
That's not the connector issue, it's normal.  
your finger have a noise to the sensor like the RF, that's reason we use the RF antenna cable.

rmbq, 2025-02-16:
Using M102 S-7 just starts a calibration: 

> 15:59:07.323 M102 S‑7
> 15:59:07.328 ok
> 15:59:07.343 G1Z0.0, Z:0.00
> 15:59:07.944 w:0, Z:0.00
> 15:59:07.955 G1Z0.1, Z:0.10
> 15:59:08.687 w:1, Z:0.10
> 15:59:08.687 wait
> 15:59:08.696 G1Z0.2, Z:0.20
> 15:59:09.427 w:2, Z:0.20
> 15:59:09.437 G1Z0.3, Z:0.30
> 15:59:10.167 w:3, Z:0.30
> 15:59:10.167 wait
> 15:59:10.178 G1Z0.4, Z:0.40
> 15:59:10.909 w:4, Z:0.40
> 15:59:10.919 G1Z0.5, Z:0.50
> 15:59:11.650 w:5, Z:0.50
> 15:59:11.650 wait
> 15:59:11.660 G1Z0.6, Z:0.60
> 15:59:12.390 w:6, Z:0.60
> 15:59:12.402 G1Z0.7, Z:0.70
> 15:59:13.132 w:7, Z:0.70
> 15:59:13.132 wait
> 15:59:13.142 G1Z0.8, Z:0.80
> 15:59:13.873 w:8, Z:0.80
> 15:59:13.883 G1Z0.9, Z:0.90
> 15:59:14.614 w:9, Z:0.90
> 15:59:14.614 wait
> 15:59:14.624 G1Z1.0, Z:1.00
> 15:59:15.355 w:10, Z:1.00
> 15:59:15.365 G1Z1.1, Z:1.10
> 15:59:16.096 w:11, Z:1.10
> 15:59:16.096 wait
> 15:59:16.106 G1Z1.2, Z:1.20
> 15:59:16.837 w:12, Z:1.20
> 15:59:16.847 G1Z1.3, Z:1.30
> 15:59:17.579 w:13, Z:1.30
> 15:59:17.579 wait
> 15:59:17.588 G1Z1.4, Z:1.40
> 15:59:18.319 w:14, Z:1.40
> 15:59:18.330 G1Z1.5, Z:1.50
> 15:59:19.061 w:15, Z:1.50
> 15:59:19.061 wait
> 15:59:19.070 G1Z1.6, Z:1.60
> 15:59:19.801 w:16, Z:1.60
> 15:59:19.812 G1Z1.7, Z:1.70
> 15:59:20.543 w:17, Z:1.70
> 15:59:20.543 wait
> 15:59:20.552 G1Z1.8, Z:1.80
> 15:59:21.283 w:18, Z:1.80
> 15:59:21.293 G1Z1.9, Z:1.90
> 15:59:22.025 w:19, Z:1.90
> 15:59:22.025 wait
> 15:59:22.034 G1Z2.0, Z:2.00
> 15:59:22.764 w:20, Z:2.00
> 15:59:22.775 G1Z2.1, Z:2.10
> 15:59:23.506 w:21, Z:2.10
> 15:59:23.506 wait
> 15:59:23.516 G1Z2.2, Z:2.20
> 15:59:24.247 w:22, Z:2.20
> 15:59:24.257 G1Z2.3, Z:2.30
> 15:59:24.988 w:23, Z:2.30
> 15:59:24.988 wait
> 15:59:24.998 G1Z2.4, Z:2.40
> 15:59:25.729 w:24, Z:2.40
> 15:59:25.740 G1Z2.5, Z:2.50
> 15:59:26.471 w:25, Z:2.50
> 15:59:26.471 wait
> 15:59:26.480 G1Z2.6, Z:2.60
> 15:59:27.211 w:26, Z:2.60
> 15:59:27.221 G1Z2.7, Z:2.70
> 15:59:27.953 w:27, Z:2.70
> 15:59:27.953 wait
> 15:59:27.962 G1Z2.8, Z:2.80
> 15:59:28.692 w:28, Z:2.80
> 15:59:28.703 G1Z2.9, Z:2.90
> 15:59:29.435 w:29, Z:2.90
> 15:59:29.435 wait
> 15:59:29.445 G1Z3.0, Z:3.00
> 15:59:30.174 w:30, Z:3.00
> 15:59:30.185 G1Z3.1, Z:3.10
> 15:59:30.917 w:31, Z:3.10
> 15:59:30.917 wait
> 15:59:30.926 G1Z3.2, Z:3.20
> 15:59:31.657 w:32, Z:3.20
> 15:59:31.668 G1Z3.3, Z:3.30
> 15:59:32.399 w:33, Z:3.30
> 15:59:32.399 wait
> 15:59:32.408 G1Z3.4, Z:3.40
> 15:59:33.140 w:34, Z:3.40
> 15:59:33.149 G1Z3.5, Z:3.50
> 15:59:33.881 w:35, Z:3.50
> 15:59:33.881 wait
> 15:59:33.890 G1Z3.6, Z:3.60
> 15:59:34.621 w:36, Z:3.60
> 15:59:34.631 G1Z3.7, Z:3.70
> 15:59:35.362 w:37, Z:3.70
> 15:59:35.362 wait
> 15:59:35.373 G1Z3.8, Z:3.80
> 15:59:36.103 w:38, Z:3.80
> 15:59:36.113 G1Z3.9, Z:3.90
> 15:59:36.844 w:39, Z:3.90
> 15:59:36.844 wait
> 15:59:36.855 BD Sensor calibrated.
> 15:59:37.842 wait
> 15:59:38.843 wait
> 15:59:39.842 wait
> 15:59:40.843 wait
> 15:59:41.842 wait

> 16:13:13.136 M102 S‑5
> 16:13:13.141 ok
> 16:13:13.270 Calibrate data:0,1015
> 16:13:13.270 Bad BD Sensor height! Recommended distance 0.5‑2.0mm
> 16:13:13.323 Calibrate data:1,1015
> 16:13:13.373 Calibrate data:2,1015
> 16:13:13.423 Calibrate data:3,1015
> 16:13:13.473 Calibrate data:4,1015
> 16:13:13.523 Calibrate data:5,1015
> 16:13:13.573 Calibrate data:6,1015
> 16:13:13.623 Calibrate data:7,1015
> 16:13:13.673 Calibrate data:8,1015
> 16:13:13.723 Calibrate data:9,1015
> 16:13:13.773 Calibrate data:10,1015
> 16:13:13.823 Calibrate data:11,1015
> 16:13:13.873 Calibrate data:12,1015
> 16:13:13.923 Calibrate data:13,1015
> 16:13:13.973 Calibrate data:14,1015
> 16:13:14.023 Calibrate data:15,1015
> 16:13:14.073 Calibrate data:16,1015
> 16:13:14.123 Calibrate data:17,1015
> 16:13:14.173 Calibrate data:18,1015
> 16:13:14.223 Calibrate data:19,1015
> 16:13:14.273 Calibrate data:20,1015
> 16:13:14.324 Calibrate data:21,1015
> 16:13:14.373 Calibrate data:22,1015
> 16:13:14.424 Calibrate data:23,1015
> 16:13:14.474 Calibrate data:24,1015
> 16:13:14.523 Calibrate data:25,1015
> 16:13:14.573 Calibrate data:26,1015
> 16:13:14.623 Calibrate data:27,1015
> 16:13:14.674 Calibrate data:28,1015
> 16:13:14.723 Calibrate data:29,1015
> 16:13:14.773 Calibrate data:30,1015
> 16:13:14.823 Calibrate data:31,1015
> 16:13:14.873 Calibrate data:32,1015
> 16:13:14.924 Calibrate data:33,1015
> 16:13:14.973 Calibrate data:34,1015
> 16:13:15.023 Calibrate data:35,1015
> 16:13:15.073 Calibrate data:36,1015
> 16:13:15.123 Calibrate data:37,1015
> 16:13:15.173 Calibrate data:38,1015
> 16:13:15.223 Calibrate data:39,1015

And all values are 1015. The Z is moving correctly from 0 to 3.9mm, attached video of z steps and the image of the gap between the bed and the nozzle at the end of the calibration 

https://github.com/user-attachments/assets/f4e503c4-da45-457e-bba7-a21fb2ee3019

[image]

markniu, 2025-02-17:
oh, I thought it was klipper, and the raw data reading gcode M102 S-7 is not for the marlin.
how about the led on the sensor?

[image]

rmbq, 2025-02-17:
Sorry, i didn't say it was marlin.
The led is always slow blinking. I tried 2 different beds (the one in the photo and another one from creality) and also to put the sensor on different metal surfaces

markniu, 2025-02-17:
It should be fast blinking, you can check again. if still in that case you can contact the seller for a replacement

> Sorry, i didn't say it was marlin. The led is always slow blinking. I tried 2 different beds (the one in the photo and another one from creality) and also to put the sensor on different metal surfaces

rmbq, 2025-02-17:
I bought it on
Pandapi3d with the same mail associated with this account. 
I wrote to markyueyue@gmail.com
Thanks

closed 2025-02-17 by rmbq