markniu/Bed_Distance_sensor #216: 2 sensors bad

mikemerryguy, 2025-02-16:
I have 2 bad sensors from the USA distributor, he told me to open an issue here, I also posted in the FB group

One sensor is completely dead, it has zero power whatsoever!
there is no troubleshooting that...

The other sensor is blinking slowly and when the Z is triggered it blinks fast.

I have 4 machines that all use the BD Sensor and this one new sensor blinks slow or fast no matter what machine I put it in.  the LED is always blinking.

Also if i lay the sensor FLAT directly on the bed, it says too far away and if i lift just one side up it shows 0.0mm 

Again this is in ALL four machines, mounts don't matter, one sensor works in the mount, this one does not and flashes like crazy!

all running klipper
two printers have skr 1.4 turbo 
two running manta e3ez boards using the SPI to CAN pins (only without filters)

markniu, 2025-02-17:
Hello, you can check again with the following 2 ways:

1. Check LED status:
     Put a metal close to the coil about 1mm. to see if the light is on or fast blinking? else it should be bad.

2. Check the data with gcode:
     Make sure you have connected the sensor to your printer and M102 S-1 works.
     Put a metal close to the coil about 0.5mm, Sending gcode M102 S-7 to get the raw data.
     Put the meta close to coil about 1mm, Sending gcode M102 S-7 to get the second raw data.
     the second data should be bigger than the first one at least 30, else it should be bad.

You can ask the seller to send the replacement to you, then I will send that replacement to he in and next time.

mikemerryguy, 2025-02-17:
like I said, 1 sensor has NO led at all!!

The second is giving random numbers every calibration!

I think the coil got mounting inside the cap at an angle, if you angle it just right, it seems to work properly.

3:21 PM 377 at 0.2mm
3:21 PM 373 at 0.1mm
3:21 PM 379 at 0.0mm
3:20 PM Raw data:308 at 0 mm, BDsensor to bed: 1.2320 mm, Bed: 20.7C

4:02 PM 364 at 0.2mm
4:02 PM 357 at 0.1mm
4:02 PM 364 at 0.0mm
4:01 PM Raw data:299 at 0 mm, BDsensor to bed: 1.1960 mm, Bed: 20.3C

4:03 PM 369 at 0.2mm
4:03 PM 364 at 0.1mm
4:03 PM 370 at 0.0mm
4:03 PM Raw data:303 at 0 mm, BDsensor to bed: 1.2120 mm, Bed: 20.4C

I have changed nothing between these calibrations!!!

closed 2025-02-24 by markniu