markniu/Bed_Distance_sensor #218: Sensorless delta setup and BDsensor collision_homing

potokslow, 2025-02-22:
Hello,

I'm refitting an old delta printer and I'd like to use BDsensor for automatic Z-offset / collision homing, while the A/B/C motors would be set up for sensorless homing.
I'm using Anycubic Kossel, BTT SKR Pico, EBB42 (on the effector), Pi 3B with recent Klipper.

BDsensor documentation calls for using `endstop_pin: probe:z_virtual_endstop` for Z stepper, but this wouldn't directly work with delta-style printer.
Delta printer homes upwards, uses three steppers for controlling Z distance, plus they already have they respective `endstop_pin: tmc2209_stepper_#:virtual_endstop` defined for sensorless homing.

Is there any way to configure automatic Z-offset via BDsensor in this case?

VoldDesigns, 2025-02-22:
Excuse my ignorance, I know nothing about delta printers. But saw your issue and figured I'd do a little research. From what I saw with other delta configs is you don't use the:
```
endstop_pin: probe:z_virtual_endstop

```
(For BLTouch, however, this leads me to believe the BD sensor would be the same) 
But in that research I found this
[video](https://youtu.be/-8XKzrWYpS4?si=I-oEMYtkdVGRItbv).

It might help you and there is some useful information the description! Best of luck!

potokslow, 2025-02-22:
Thank you for your suggestion @VoldDesigns.
This is my first delta and I'm plenty ignorant myself, I'm here to learn ;)

My case is a bit different than the setup in the video you shared.
The printer in the video is using generic endstop switches for sensor-based homing, while I'm trying to use sensorless homing.

When talking about the differences between typical cartesian printer and delta, it's mainly about how the motors handle the motion and the direction of Z homing.
In cartesian printer, X, Y and Z motors are used, each responsible for motion on their own axis - e.g. you only use motor X to move in X axis. Z axis is typically homed towards the print bed.
In delta printer, motors A, B and C are configured, each handling motion on their own vertical tower. Motors work together to move the tool head / effector- e.g. you'd need to use all motors to achieve reliable motion on any axis. All motors home upwards, away from print bed.
For sensorless homing on cartesian printer, the `virtual_endstop` for the respective axis is used, e.g. `endstop_pin: probe:z_virtual_endstop` for Z axis.
In case of delta sensorless homing, virtual endstop needs to be configured for all axes, i.e.:

> [stepper_a]
> endstop_pin: tmc2209_stepper_a:virtual_endstop  
> [stepper_b]  
> endstop_pin: tmc2209_stepper_b:virtual_endstop  
> [stepper_c]  
> endstop_pin: tmc2209_stepper_c:virtual_endstop

In my case, with the the delta printer:
- All steppers will be configured with virtual endstops for sensorless homing.
- Since this is delta configuration, there isn't even the Z stepper on which the `endstop_pin: probe:z_virtual_endstop` from BDsensor docs should be configured.

I think that using BDsensor as a bed probe could be done with the generic `[probe]` Klipper section.
The question here is the using BDsensor not like a simple probe, but also for automatic Z-offset / collision homing.
Can this be done on a delta?