markniu/Bed_Distance_sensor #219: Real Time leveling

Val-dose-things, 2025-02-25:
G28
BDSENSOR_SET REAL_TIME_HEIGHT=0.3
G1 Z0.2

results in:

[image]

[image]

[klippy.log](https://github.com/user-attachments/files/18970857/klippy.log)

i believe its and issue with it triggering to protect it from crashing any ideas?
everything else is working great.

markniu, 2025-02-26:
this is because the sensor and the z motors are not connected in the same MCU. it's not possible to do real time adjust from the high level python between different MCU during printing. 
so you need to change the connection from the extra mcu to the mcu of z motors.

Val-dose-things, 2025-02-26:
ahhhh ok that would make sense. this is what i get for trying to be fancy and us the extra mcu bltouch connection.