markniu/Bed_Distance_sensor #222: Printer can'r boot with DGUS_LCD_UI IA_CREALITY

rmbq, 2025-03-05:
I followed all the instructions to add support to my marlin for my printer.
My source tree is here, branch rmbq-2.1.x:
https://github.com/rmbq/Marlin
and the modifications to support BD sensor are here:
https://github.com/rmbq/Marlin/commit/f263a99a4dc58d2ea9ce2c809f0e213b35bfaee1

I tried with and without PR https://github.com/MarlinFirmware/Marlin/pull/27243

Disabling AI_CREALITY and EXTENDED_UI the printer boots fine

I think everyone can trigger this issue enabling AI_CREALITY and EXTENDED_UI.

Do you have any suggestion to help me debug this issue? a way to add more debug prints or something else?

Thanks

markniu, 2025-03-05:
Check the size of the bin file, the flash address of the mcu, offset of the EEPROM.
how about disable the EEPROM ?

rmbq, 2025-03-05:
Seems to be a Marlin issue

closed 2025-03-05 by rmbq