markniu/Bed_Distance_sensor #225: Internal error on command:"BED_MESH_CALIBRATE" --- Internal error in stepcompress

Qtronics-Electronics, 2025-03-15:
This is a new build and I keep getting The following error when I try to run Bed Mesh calibrate. I updated Klipper, and now I am not sure what to try next. 

File "/home/pi/klipper/klippy/stepper.py", line 230, in generate_steps
    raise error("Internal error in stepcompress")

`[BDsensor] 
# Don't use aliases for the board pins
sda_pin: EBBCan:PB8
scl_pin: EBBCan:PB9
delay: 20 # you can set it 10 if the BDsensor version is >=1.2
z_offset:0 #  within -0.6 to 0.6mm
x_offset: -34
y_offset: 0
no_stop_probe: # fast probe that the toolhead will not stop at the probe point,disable it by commenting out.
position_endstop: 2 #the triggered position, recommend value is 1~2.8
collision_homing:1 #  set it 1 to enable homing with nozzle collision sensing.
collision_calibrate:1 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.

#QGL_Tilt_Probe:0 #set 1 to enable probe up and down when do quad_gantry_level

`[bed_mesh]
speed: 60
probe_count: 7, 7
horizontal_move_z: 10
algorithm: bicubic # Options bicubic or lagrange
mesh_min : 10, 10
mesh_max : 250, 250 
mesh_pps: 4, 4
zero_reference_position: 10, 10
bicubic_tension: 0.2`

[klippy.log](https://github.com/user-attachments/files/19262934/klippy.log)

markniu, 2025-03-19:
sorry for replay late,
Here is the error in the klippy.log file
```
Unhandled exception during run
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 176, in run
    self.reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 292, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 340, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/pi/klipper/klippy/reactor.py", line 158, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/pi/klipper/klippy/extras/BDsensor.py", line 336, in scan_sample_event
    intd = self.mcu_probe.BD_Sensor_Read(0)
  File "/home/pi/klipper/klippy/extras/BDsensor.py", line 857, in BD_Sensor_Read
    raise self.printer.command_error("Bed Distance Sensor, "
gcode.CommandError: Bed Distance Sensor, out of range.:10.15 
```

I find the `horizontal_move_z:10 ` in the bed_mesh section is too high, please set it to 1

mikemerryguy, 2025-03-20:
i had this issue make sure you have file manager on in the mooonraker.conf

[file_manager]
enable_object_processing: True

make sure you have exclude objects and M486 in printer.cfg and exclude object

[exclude_object] 
[gcode_macro M486]
description: exlude objects in UI as print is running, must enable object label in slicer
gcode:
  # Parameters known to M486 are as follows:
  #   [C<flag>] Cancel the current object
  #   [P<index>] Cancel the object with the given index
  #   [S<index>] Set the index of the current object.
  #       If the object with the given index has been canceled, this will cause
  #       the firmware to skip to the next object. The value -1 is used to
  #       indicate something that isn’t an object and shouldn’t be skipped.
  #   [T<count>] Reset the state and set the number of objects
  #   [U<index>] Un-cancel the object with the given index. This command will be
  #       ignored if the object has already been skipped

  {% if 'exclude_object' not in printer %}
    {action_raise_error("[exclude_object] is not enabled")}
  {% endif %}

  {% if 'T' in params %}
    EXCLUDE_OBJECT RESET=1

    {% for i in range(params.T | int) %}
      EXCLUDE_OBJECT_DEFINE NAME={i}
    {% endfor %}
  {% endif %}

  {% if 'C' in params %}
    EXCLUDE_OBJECT CURRENT=1
  {% endif %}

  {% if 'P' in params %}
    EXCLUDE_OBJECT NAME={params.P}
  {% endif %}

  {% if 'S' in params %}
    {% if params.S == '-1' %}
      {% if printer.exclude_object.current_object %}
        EXCLUDE_OBJECT_END NAME={printer.exclude_object.current_object}
      {% endif %}
    {% else %}
      EXCLUDE_OBJECT_START NAME={params.S}
    {% endif %}
  {% endif %}

  {% if 'U' in params %}
    EXCLUDE_OBJECT RESET=1 NAME={params.U}
  {% endif %}

also make sure your slicer is set to label objects and klipper as slicer