markniu/Bed_Distance_sensor #226: I am having an issue with z hop

mikemerryguy, 2025-03-20:
On bed mesh calibrate my Z keeps doing a HOP, my other 3 printers scan the bed without the hop, I am sure I am missing something, looking for another set of eyes to take a peek at my CFG, this is a new printer and isn't done, running a old Sunlu S8 with a Manta e3 EZ board and usb to serial FDTI cable (uart mode) all the axis and runout sensor work find, I have changed all the motors with Creality version motors and running a CR10 stock (ender 3) lcd, its running on a bowden setup as well.

Since I am using a FDTI cable I used the SPI pins for the bdsensor as the bltouch pins on this board are filtered and will not work! 

`[include mainsail.cfg]
[include adv_filament_swap.cfg]
[include tones.cfg]
#[include adx.cfg] #for input shaping only

[input_shaper]
shaper_freq_x: 72.4 #15400
shaper_type_x: mzv 
shaper_freq_y: 35.8 #<= 3800 mm/sec^2
shaper_type_y: mzv

[safe_z_home]
home_xy_position: 199,158 # changes based on probe offsets
speed: 150
z_hop: 5
z_hop_speed: 5

[BDsensor] 
# Don't use aliases for the board pins
sda_pin: PC12  #white WIRE
scl_pin: PC11  #yellow WIRE
delay: 8 # you can set it 10 if the BDsensor version is >=1.2
#z_offset: 0 #  within -0.6 to 0.6mm
x_offset: -44
y_offset: -3
no_stop_probe: # fast probe that the toolhead will not stop at the probe point,disable it by commenting out.
position_endstop: 1.2 #the triggered position, recommend value is 1~2.8
collision_homing: 1 #  set it 1 to enable homing with nozzle collision sensing.
collision_calibrate: 1 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.
speed: 2
QGL_Tilt_Probe: 0 #set 1 to enable probe up and down when do quad_gantry_level
homing_cmd: G990028
  
  
[gcode_macro _CLIENT_VARIABLE]
variable_use_custom_pos   : True ; use custom park coordinates for x,y [True/False]
variable_custom_park_x    : 0   ; custom x position; value must be within your defined min and max of X
variable_custom_park_y    : 0   ; custom y position; value must be within your defined min and max of Y
variable_custom_park_dz   : 2.0   ; custom dz value; the value in mm to lift the nozzle when move to park position
variable_retract          : 6.0   ; the value to retract while PAUSE
variable_cancel_retract   : 6.0   ; the value to retract while CANCEL_PRINT
variable_speed_retract    : 25.0  ; retract speed in mm/s
variable_unretract        : 6.0   ; the value to unretract while RESUME
variable_speed_unretract  : 25.0  ; unretract speed in mm/s
variable_speed_hop        : 15.0  ; z move speed in mm/s
variable_speed_move       : 250.0 ; move speed in mm/s
variable_park_at_cancel   : True ; allow to move the toolhead to park while execute CANCEL_PRINT [True/False]
variable_park_at_cancel_x : 0  ; different park position during CANCEL_PRINT [None/Position as Float]; park_at_cancel must be True
variable_park_at_cancel_y : 235  ; different park position during CANCEL_PRINT [None/Position as Float]; park_at_cancel must be True
## !!! Caution [firmware_retraction] must be defined in the printer.cfg if you set use_fw_retract: True !!!
#variable_use_fw_retract   : False ; use fw_retraction instead of the manual version [True/False]
variable_idle_timeout     : 43200 ; time in sec until idle_timeout kicks in. Value 0 means that no value will be set or restored
#variable_runout_sensor    : ""    ; If a sensor is defined, it will be used to cancel the execution of RESUME in case no filament is detected.
##                                   Specify the config name of the runout sensor e.g "filament_switch_sensor runout". Hint use the same as in your printer.cfg
## !!! Custom macros, please use with care and review the section of the corresponding macro.
## These macros are for simple operations like setting a status LED. Please make sure your macro does not interfere with the basic macro functions.
## Only  single line commands are supported, please create a macro if you need more than one command.
#variable_user_pause_macro : ""    ; Everything inside the "" will be executed after the klipper base pause (PAUSE_BASE) function
#variable_user_resume_macro: ""    ; Everything inside the "" will be executed before the klipper base resume (RESUME_BASE) function
#variable_user_cancel_macro: ""    ; Everything inside the "" will be executed before the klipper base cancel (CANCEL_PRINT_BASE) function
gcode:

[gcode_macro BED_MESH_CALIBRATE]
rename_existing: BED_MESH_CALI
gcode:
  Z_TILT_ADJUST
[... 662 lines cut ...]
#*# 	-0.288333, -0.325000, -0.385000, -0.377500, -0.367500, -0.365000, -0.345000, -0.322500, -0.325000, -0.248333, -0.227500, -0.165000
#*# 	-0.183333, -0.195000, -0.265000, -0.270833, -0.258333, -0.297500, -0.275000, -0.249167, -0.267500, -0.197500, -0.195000, -0.142500
#*# 	-0.085000, -0.100000, -0.160000, -0.179167, -0.190000, -0.205000, -0.202500, -0.195000, -0.223333, -0.180000, -0.160000, -0.105000
#*# 	-0.050833, -0.052500, -0.117500, -0.122500, -0.125833, -0.182500, -0.165000, -0.160833, -0.195000, -0.155000, -0.165000, -0.105000
#*# 	0.010000, -0.002500, -0.055833, -0.070000, -0.085000, -0.122500, -0.125000, -0.145000, -0.191667, -0.157500, -0.175000, -0.103333
#*# 	0.065000, 0.071667, 0.021667, 0.012500, -0.007500, -0.043333, -0.063333, -0.095000, -0.145000, -0.110000, -0.125000, -0.085000
#*# 	0.095000, 0.091667, 0.055000, 0.045000, 0.032500, -0.008333, -0.045000, -0.070000, -0.125000, -0.118333, -0.130000, -0.075000
#*# 	0.177500, 0.191667, 0.141667, 0.132500, 0.123333, 0.060000, 0.032500, -0.013333, -0.072500, -0.057500, -0.087500, -0.062500
#*# 	0.125000, 0.165000, 0.123333, 0.125000, 0.111667, 0.057500, 0.029167, -0.002500, -0.092500, -0.085000, -0.120833, -0.087500
#*# 	0.107500, 0.166667, 0.140000, 0.135000, 0.122500, 0.077500, 0.045000, 0.010000, -0.067500, -0.052500, -0.105000, -0.080833
#*# 	0.055000, 0.142500, 0.135000, 0.155000, 0.145000, 0.111667, 0.080000, 0.025000, -0.050000, -0.050000, -0.101667, -0.090833
#*# x_count = 12
#*# y_count = 12
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = bicubic
#*# tension = 0.2
#*# min_x = 25.0
#*# max_x = 292.96
#*# min_y = 25.0
#*# max_y = 286.91
#*#
#*# [BDsensor]
#*# z_offset = 0
`

markniu, 2025-03-20:
I think you should remove all the `METHOD=rapid_scan` because it is only for the BTT eddy sensor.

mikemerryguy, 2025-03-20:
> I think you should remove all the `METHOD=rapid_scan` because it is only for the BTT eddy sensor.

still hopping!!!
```

[include mainsail.cfg]
[include adv_filament_swap.cfg]
[include tones.cfg]
#[include adx.cfg] #for input shaping only

[input_shaper]
shaper_freq_x: 72.4 #15400
shaper_type_x: mzv 
shaper_freq_y: 35.8 #<= 3800 mm/sec^2
shaper_type_y: mzv

[safe_z_home]
home_xy_position: 199,158 # changes based on probe offsets
speed: 150
z_hop: 5
z_hop_speed: 5

[BDsensor] 
# Don't use aliases for the board pins
sda_pin: PC12  #white WIRE
scl_pin: PC11  #yellow WIRE
delay: 8 # you can set it 10 if the BDsensor version is >=1.2
#z_offset: 0 #  within -0.6 to 0.6mm
x_offset: -44
y_offset: -3
no_stop_probe: # fast probe that the toolhead will not stop at the probe point,disable it by commenting out.
position_endstop: 1.2 #the triggered position, recommend value is 1~2.8
collision_homing: 1 #  set it 1 to enable homing with nozzle collision sensing.
collision_calibrate: 1 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.
speed: 2
QGL_Tilt_Probe: 0 #set 1 to enable probe up and down when do quad_gantry_level
homing_cmd: G990028
  
  
[gcode_macro _CLIENT_VARIABLE]
variable_use_custom_pos   : True ; use custom park coordinates for x,y [True/False]
variable_custom_park_x    : 0   ; custom x position; value must be within your defined min and max of X
variable_custom_park_y    : 0   ; custom y position; value must be within your defined min and max of Y
variable_custom_park_dz   : 2.0   ; custom dz value; the value in mm to lift the nozzle when move to park position
variable_retract          : 6.0   ; the value to retract while PAUSE
variable_cancel_retract   : 6.0   ; the value to retract while CANCEL_PRINT
variable_speed_retract    : 25.0  ; retract speed in mm/s
variable_unretract        : 6.0   ; the value to unretract while RESUME
variable_speed_unretract  : 25.0  ; unretract speed in mm/s
variable_speed_hop        : 15.0  ; z move speed in mm/s
variable_speed_move       : 250.0 ; move speed in mm/s
variable_park_at_cancel   : True ; allow to move the toolhead to park while execute CANCEL_PRINT [True/False]
variable_park_at_cancel_x : 0  ; different park position during CANCEL_PRINT [None/Position as Float]; park_at_cancel must be True
variable_park_at_cancel_y : 235  ; different park position during CANCEL_PRINT [None/Position as Float]; park_at_cancel must be True
## !!! Caution [firmware_retraction] must be defined in the printer.cfg if you set use_fw_retract: True !!!
#variable_use_fw_retract   : False ; use fw_retraction instead of the manual version [True/False]
variable_idle_timeout     : 43200 ; time in sec until idle_timeout kicks in. Value 0 means that no value will be set or restored
#variable_runout_sensor    : ""    ; If a sensor is defined, it will be used to cancel the execution of RESUME in case no filament is detected.
##                                   Specify the config name of the runout sensor e.g "filament_switch_sensor runout". Hint use the same as in your printer.cfg
## !!! Custom macros, please use with care and review the section of the corresponding macro.
## These macros are for simple operations like setting a status LED. Please make sure your macro does not interfere with the basic macro functions.
## Only  single line commands are supported, please create a macro if you need more than one command.
#variable_user_pause_macro : ""    ; Everything inside the "" will be executed after the klipper base pause (PAUSE_BASE) function
#variable_user_resume_macro: ""    ; Everything inside the "" will be executed before the klipper base resume (RESUME_BASE) function
#variable_user_cancel_macro: ""    ; Everything inside the "" will be executed before the klipper base cancel (CANCEL_PRINT_BASE) function
gcode:

[gcode_macro BED_MESH_CALIBRATE]
rename_existing: BED_MESH_CALI
gcode:
[... 674 lines cut ...]
#*# 	-0.183333, -0.195000, -0.265000, -0.270833, -0.258333, -0.297500, -0.275000, -0.249167, -0.267500, -0.197500, -0.195000, -0.142500
#*# 	-0.085000, -0.100000, -0.160000, -0.179167, -0.190000, -0.205000, -0.202500, -0.195000, -0.223333, -0.180000, -0.160000, -0.105000
#*# 	-0.050833, -0.052500, -0.117500, -0.122500, -0.125833, -0.182500, -0.165000, -0.160833, -0.195000, -0.155000, -0.165000, -0.105000
#*# 	0.010000, -0.002500, -0.055833, -0.070000, -0.085000, -0.122500, -0.125000, -0.145000, -0.191667, -0.157500, -0.175000, -0.103333
#*# 	0.065000, 0.071667, 0.021667, 0.012500, -0.007500, -0.043333, -0.063333, -0.095000, -0.145000, -0.110000, -0.125000, -0.085000
#*# 	0.095000, 0.091667, 0.055000, 0.045000, 0.032500, -0.008333, -0.045000, -0.070000, -0.125000, -0.118333, -0.130000, -0.075000
#*# 	0.177500, 0.191667, 0.141667, 0.132500, 0.123333, 0.060000, 0.032500, -0.013333, -0.072500, -0.057500, -0.087500, -0.062500
#*# 	0.125000, 0.165000, 0.123333, 0.125000, 0.111667, 0.057500, 0.029167, -0.002500, -0.092500, -0.085000, -0.120833, -0.087500
#*# 	0.107500, 0.166667, 0.140000, 0.135000, 0.122500, 0.077500, 0.045000, 0.010000, -0.067500, -0.052500, -0.105000, -0.080833
#*# 	0.055000, 0.142500, 0.135000, 0.155000, 0.145000, 0.111667, 0.080000, 0.025000, -0.050000, -0.050000, -0.101667, -0.090833
#*# x_count = 12
#*# y_count = 12
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = bicubic
#*# tension = 0.2
#*# min_x = 25.0
#*# max_x = 292.96
#*# min_y = 25.0
#*# max_y = 286.91
#*#
#*# [BDsensor]
#*# z_offset = 0

```

mikemerryguy, 2025-03-20:
changed my bed mesh section to this

```
[bed_mesh]
speed: 120
horizontal_move_z: 0
mesh_min: 25, 25
mesh_max: 291, 286
probe_count: 12, 12
mesh_pps: 2, 2
algorithm: bicubic
bicubic_tension: 0.2
zero_reference_position: 159, 158
 
```

but getting Probe triggered prior to movement as soon as it starts now!!!

markniu, 2025-03-20:
you can attache the klippy.log file here and take a short video about the z_hop would be better.

mikemerryguy, 2025-03-20:
> How about remove the `[gcode_macro BED_MESH_CALIBRATE] ` and replace the other BED_MESH_CALI to BED_MESH_CALIBRATE ?

I can try tomorrow
But right now the bed_mesh_cali is defaut klipper command and only called in start leint

The macro simply auto homes before running mesh I removed the z tilt as it is slow and i have been testing over and over! Lol

I think i am getting closer i changed z trigger to .5 and z hop to 1 or close in that range i think i need to calibrate the z trigger somehow or maybe sensor is too close to bed / nozzle