markniu/Bed_Distance_sensor #227: BD Sensor stopped detecting metal, previously inconsistent Z-offset

grom79, 2025-03-20:
Each print requires for the z-offset to be adjusted. 

Tried running in collision mode, made the nozzle press into the bed and twist the toolhead a bit before probe detected. Had to recalibrate after each homing, and each homing made the z-offset wildly inaccurate, +/- 0.4mm, to the point it was either rubbing the nozzle into the bed or printing in air and required large adjustments. 

In non-collision mode, it still required baby-stepping, but of lesser, with sensor being 1mm higher than the nozzle. First calibration detected 0=100, later calibrations showed values around 200 without any changes to the probe position nor the nozzle, with nozzle being cleaned before each calibration and checked. 

After last print, BD sensor doesn't recognize metal at all, even checking with metal spatula it's not detecting, LED is slowly blinking and command for detecting probe returns normal answer, yet commands for detecting distance state "too close/far". 

[BDsensor] 
# Don't use aliases for the board pins
sda_pin: PC14
scl_pin: PA1
delay: 10 # you can set it 10 if the BDsensor version is >=1.2
#z_offset:0.0 #  within -0.6 to 0.6mm
x_offset: 0
y_offset: 25
no_stop_probe: # fast probe that the toolhead will not stop at the probe point,disable it by commenting out.
position_endstop: 1.2 #the triggered position, recommend value is 1~2.8
collision_homing:0 #  set it 1 to enable homing with nozzle collision sensing.
collision_calibrate:0 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.
#QGL_Tilt_Probe:0 #set 1 to enable probe up and down when do quad_gantry_level
speed:3 # this speed only works for the z tilt and PROBE_ACCURACY command.

[safe_z_home]
home_xy_position: 120, 120
z_hop: 5 

[bed_mesh]
speed: 500
horizontal_move_z:1 # 0.7~1.0mm is recommended
algorithm: bicubic  # this should be bicubic if the bed mesh points count is > 6*6
zero_reference_position: 120, 120 # Set this value to be the same as home_xy_position that is in the section safe_z_home
mesh_min: 10,23
mesh_max: 220,211
probe_count: 25
scan_overshoot: 8

markniu, 2025-03-21:
what's the value after M102 S-5 ? you can paste all here. also the [stepper_z] 

> First calibration detected 0=100, later calibrations showed values around 200 without any changes to the probe position nor the nozzle, with nozzle being cleaned before each calibration and checked.

grom79, 2025-03-21:
For M102 S-5:

```
952 at 3.9mm
944 at 3.8mm
937 at 3.7mm
928 at 3.6mm
919 at 3.5mm
907 at 3.4mm
895 at 3.3mm
884 at 3.2mm
873 at 3.1mm
861 at 3.0mm
847 at 2.9mm
832 at 2.8mm
818 at 2.7mm
804 at 2.6mm
787 at 2.5mm
767 at 2.4mm
750 at 2.3mm
732 at 2.2mm
713 at 2.1mm
691 at 2.0mm
671 at 1.9mm
648 at 1.8mm
625 at 1.7mm
602 at 1.6mm
576 at 1.5mm
552 at 1.4mm
526 at 1.3mm
498 at 1.2mm
470 at 1.1mm
441 at 1.0mm
413 at 0.9mm
384 at 0.8mm
357 at 0.7mm
326 at 0.6mm
299 at 0.5mm
269 at 0.4mm
241 at 0.3mm
215 at 0.2mm
188 at 0.1mm
175 at 0.0mm
```

For [stepper_z]:

```
##################################################################################
# Z Motion Setup
##################################################################################

[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
microsteps: 16
rotation_distance: 3.2
#gear_ratio: 80:16
#full_steps_per_rotation: 200 #200 for 1.8 degree, 400 for 0.9 degree

# Endstop / homing
endstop_pin: probe:z_virtual_endstop
#position_endstop: 0.0
position_max: 245
position_min: -5
homing_speed: 5
second_homing_speed: 3  #set this to 3 if homing with collision
homing_retract_speed: 2

[tmc2209 stepper_z]
uart_pin: PC11
tx_pin: PC10
uart_address: 1
run_current: 0.580
stealthchop_threshold: 999999
```

markniu, 2025-03-21:
the calibration data is good not  ` later calibrations showed values around 200 without any changes to the probe position nor the nozzle `  so I think the sensor is good too.
you can move the nozzle close to the bed like 0.5mm, 1mm,1.5mm and get the distance and raw data with `M102 S-2`  `M102 S-7` to see if the data is right.

markniu, 2025-03-21:
and not recommend to set the speed too high like 500 while scan, we should also consider there are some middle software in klipper.
```
[bed_mesh]
speed: 500
```

grom79, 2025-03-21:
Unfortunately it's not the case anymore. Right now it's not detecting the plate at all, even with nozzle pressed against the bed and lifted bit by bit, the console gives out: 
```
11:42
BDsensor:Out of measure Range or too close to the bed
11:42
M102 S-2
11:42
BDsensor:Out of measure Range or too close to the bed
11:42
M102 S-2
11:40
BDsensor:Out of measure Range or too close to the bed
11:40
M102 S-2
```

[image]

[image]

[image]

markniu, 2025-03-21:
How about the m102 S-7  ?
in this case l suspect their is a connection problem between the coil and pcb inside the sensor. you can check it by open the sensor case and if possible measure the small cable with multimeter

grom79, 2025-03-21:
With nozzle pressed against the bed: 
```
12:13
Raw data:1015,  Then we can calculate the distance by comparing to the calibration data:9.551mm
12:13
m102 S-7
```

I'll try to open the black sensor

grom79, 2025-05-03:
May I ask for guidance? How to open up the casing without cutting it apart?

markniu, 2025-05-04:
You can open it like this way: https://youtube.com/shorts/UNoegz9WOz0

> May I ask for guidance? How to open up the casing without cutting it apart?

markniu, 2025-05-09:
The multimeter should beep when measuring these 2 point
[image]

grom79, 2025-05-09:
Unfortunately, it seems there's no signal: https://www.youtube.com/watch?v=C_-VsUjRNGk

grom79, 2025-06-05:
Is there a way to repair this part? If not, is there an option of buying it, or would I have to buy the whole BD sensor?

markniu, 2025-06-06:
I just added the coil option to the product page for free, but you need to pay for shipping.
  https://www.pandapi3d.com/product-page/bed-distance-sensor

grom79, 2025-06-26:
I've received the new coil, but wasn't able to test it yet as my bed stopped working properly at the start of the week. I'll update when I'll be able to test the replacement coil.