markniu/Bed_Distance_sensor #229: BD Sensor not working with SKR mini e3 V3

Atreta, 2025-04-27:
I got very interested about this sensor and i just bought one bd sensor and i'm having difficulties using it on a SKR e3 mini v3. 
I compiled my own firmware following the instructions from the github., but when i try to make an auto home it moves the x and y axis as it should, but when it tries the Z axis it raises it by about 1mm and stops. 
I tried with the z end stop removed from the board and also installed and the behaviour is the same.
When i try to calibrate i get read error: check connection and delay. It isn't showing the "bd: xx" distance also. only shows the printer is ready.
 If i flash the firmware with bd sensor and all other changes for it disabled, the auto home works normally.
Thanks in advance.

markniu, 2025-04-28:
Which pins is the bdsensor connected to? it seems like the board can't communicate with the BDsensor.

Atreta, 2025-04-28:
Thanks for the reply, it's connected to PC14 and to PA1 on the z probe socket.

markniu, 2025-04-28:
As a test, you can try the pins on the exp1 connector. and use the M102 S-1 to check the connection first.

Atreta, 2025-04-28:
I'm sorry, i think I'm wasting your time. I thought the mellow one was the same as yours. I'll try the exp1, but I think this one is broken as it is getting kinda hot while plugged in, not burning hot, but hot anyway. 
Thank you for your time Mark.

Atreta, 2025-05-01:
I switched the pins to the EXP1 socket, now the printer is moving the Z axis down as it would when doing an auto home, but it halts the printer because the probe isn't detected. The calibrating program  still won't detect it. Also the probe isn't overheating  anymore.
how should i edit the pins?i'm using PB9 as sda and PD6 as clock, is it right below?

#define  I2C_BD_SDA_PIN    PB9  // Please change to the actual number which the SDA wire is connected to your mainboard
#define  I2C_BD_SCL_PIN    PD6   // Please change to the actual number which the SLK wire is connected to your mainboard
#define  I2C_BD_DELAY  20      // default value is 20, should be in the range [10,30].

thanks a lot.

markniu, 2025-05-01:
Did the bdsensor return the version string after M102 S-1? 
and I can not find there has PD6 in the board schematic

[image]

Atreta, 2025-05-01:
No, it didn't show the version yet.
This is the the schematic for exp1 on the skr mini e3 v3

[image]

markniu, 2025-05-01:
another way is to increase the delay value, for example #define I2C_BD_DELAY 30

Atreta, 2025-05-03:
I tried to use the tft pins and it didn't work, after that i tried the I/O pins and it behaved the same way when I tried to calibrate. It shows an empty bd version and the "read error check connection and delay" message.

suroh1994, 2025-06-20:
@Atreta What firmware are you running? I have a SKR mini E3 V3 with the bd sensor connected to pins PB8 and PD6. In case it helps, here is my BDsensor config segment

```
[BDsensor] 
sda_pin: PB8
scl_pin: PD6
delay: 10
z_offset: 0
x_offset: 0
y_offset: 24
no_stop_probe: 
position_endstop: 1.38
collision_homing: 1 
collision_calibrate: 1
speed: 3
```

xMikyx, 2025-07-14:
@markniu it doesn't work. I also tried on exp1. and everywhere I get the same result "No data or corrupt data from BDsensor( ), Please check connection". Did I connect the cables wrong, which I don't think I did because I followed the markings on the probe. I also tried changing the cables and the same problem. On the M119 command the probe is always with:TRIGGERED. Is there a fault in the probe?

markniu, 2025-07-14:
How about the LED on the sensor? 
I think the sensor should not be fault if the led is work as expected and the case of the sensor is not hot 

[image]

xMikyx, 2025-07-14:
@suroh1994 It is not necessary that I have a screen, I have it on Wifi and I control it on Orca. I also tried on exp1 and the same result. But this probe has two GND connections. Where can I connect the second GND on exp1. I got the same result that there is no communication. I took the BD sensor because it says that you normally connect it to the bl port.

I can try again on exp1 and let's see. That is, SDA to PB8 and CLK to PD6, GND and 5V are clear to me, but where did you connect the 2 GNDs? To which connector?

[image]

xMikyx, 2025-07-14:
@suroh1994 Now I'm completely confused :). Namely, I watched the sensor installation on the Armor3d Printing YouTube channel. I think he uses the BDsensor version M which he wanted to connect to the exp1 output and it didn't work for him. That is, he connected it to the Z-probe output and it worked for him. That's why I specifically bought a probe intended for the Z-probe output which is supposed to work on the BLtouch connectors but it doesn't work on this output and I have to try it on exp1. I also tried it on exp1 by connecting one GND to the z-probe output. I will try to connect the GND even though it doesn't make sense to me if I can, because then it would probably already be connected and we wouldn't have 5 cables. Wrong world

Armor3d Printing BD sensor install:
https://www.youtube.com/watch?v=tTFS4IEcP1s

suroh1994, 2025-07-14:
I don't know whether the is a meaningful difference between the Probe Form and the m Form. I thought it was only the Form Factor. I don't have a Probe Version, but I do have 2 m Versions running in two different printers with different Motherboards (e3 v3 and an Octopus Pro v1.1) and both Just worked. But I am using Klipper and from what i can Tell your're using Something elss (maybe marlin?) so I don't Know whether there are any Software limitations maybe?

xMikyx, 2025-07-14:
@suroh1994 I'm using klipper, everything is freshly downloaded. I also followed the steps from this armo3d in case I missed something and it didn't work. I'll try again and if I manage to fix it, I'll let you know.