markniu/Bed_Distance_sensor #236: Z-Offset too close with contact / collision homing activated

suroh1994, 2025-07-17:
I'm facing a bad z-offset on both of my printers using the bdsensor in collision homing and calibration mode.
(The parts were printed on a PEO sheet, so ignore the strip-pattern on the bottom)

[image]

As you can tell, the first layer walls look fine, but the center seems way to close, causing all these little gaps. I have this same issue on both printers, making me believe I either did something wrong on both of them or the collision homing is too close. 

Another idea was that I may be printing too fast, so I slowed down my first layer to 30mm/s for walls and 40mm/s for infill (instead of 50mm/s for walls and 105mm/s for infill) and the result looks different, but still not perfect.

[image]

Here are my relevant klipper config segments:
```
[BDsensor] 
sda_pin: PB6
scl_pin: PB7
delay: 10
z_offset: 0.0
x_offset: 0
y_offset: 35
no_stop_probe:
position_endstop: 1.38
collision_homing: 1
collision_calibrate: 1
speed: 3
homing_cmd: G28

[stepper_z]
step_pin: PF11
dir_pin: !PG3
enable_pin: !PG5
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
homing_speed: 5
second_homing_speed: 3
homing_retract_speed: 2
homing_retract_dist:5
position_min: -6
position_max: 200
```

I've tried varying the `z_offset` but that didn't seem to have an effect.

markniu, 2025-07-18:
How about adjust the z_offset while printing? 
if still work you can  try to setting the first layer height in the slicer for example: 0.2-->0.25mm
 I think you can also try to use the real time leveling that I always used: https://pandapi3d.cn/en/bdsensor/RunRealTimeLevel

suroh1994, 2025-07-18:
> How about adjust the z_offset while printing? 

I guess that could work but i would prefer a fire and forget approach, so I don't have to keep an eye on my printers, everytime I start a new print 

> if still doesn't work you can try to setting the first layer height in the slicer for example: 0.2-->0.25mm

I'm still in the process of calibrating the second printer, so I'll try that and see if it works.

> I think you can also try to use the real time leveling that I always used: https://pandapi3d.cn/en/bdsensor/RunRealTimeLevel

Interesting. My printer hast a large print bed (>350mm in both directions) and it strongly warps when heated, which is why I've enabled `move_check_distance` in the bed mesh to avoid crashing into the bed. Does the real time leveling account for long moves and perform a similar compensation?